thonny/thonny #3069: Keyboard shortcuts is confusing on windows

chenyulue, 2024-01-14:
My OS is Win11, and thonny has a confusing keyboard shortcuts. When I press key "r", a Find-and-Replace dialog pop up just as the Ctrl+F is pressed，and when I press key "t", the script is run just as pressing F5.

Are there any ways to check the default behavior for key pressing? It's annoying when typing code to pop up the Find-and-Replace dialog.

aivarannamaa, 2024-01-14:
Thonny is not supposed to behave like this and it does not on my computer. Can your keyboard have some issues?

chenyulue, 2024-01-15:
@aivarannamaa There is no problem for my keyboard. Actually the same issue happens in IDLE. Sorry not to give more details: the input method that I use is Chinese IME. I find that everything works well if I change to English IME. Maybe the problem is related to the Chinese IME.

aivarannamaa, 2024-01-15:
If this also happens in IDLE (which uses same GUI toolkit as Thonny), it's worth reporting it at https://github.com/python/cpython/issues. IDLE has more developers and they are probably more knowledgeable about this issue. Perhaps they can change something in Tkinter code, which benefits Thonny as well.