thonny/thonny #3071: Thonny locks up

AshDini, 2024-01-16:
If you don't press "Stop" after a syntax error, correct the error and press 'Run' Thonny will lock up

aivarannamaa, 2024-01-16:
What do you precisely mean by locking up? No click or keypress does anything? Can you send me a screenshot of this state?

AshDini, 2024-01-17:
Here  is the screen BEFORE:

[image: 2024-01-17 15 15 15.jpg]

and AFTER, takes no input from keyboard:

[image: 2024-01-17 15 09 12.jpg]

AshDini, 2024-01-17:
This also happens if the source is being edited while it is running.

aivarannamaa, 2024-01-17:
The pictures don't come through when sent via e-mail. Please post them at GitHub

AshDini, 2024-01-18:
[image]
[image]

I am using Thonny 4.1.4 under Win 10.
I was debugging a Python web server and while it was still running, I made some code changes, without pressing the Stop button.
In this case Thonny becomes unresposive, and has to be deleted and reloaded.

closed 2024-01-18 by AshDini