thonny/thonny #3073: Would be nice to be able to download to computer all the files on Micropython device at once

programaty, 2024-01-16:
Well, I have a Raspberry Pi Pico and sometimes I "download" all the files in int to the pc to make backups. But it's tedious to be right clicking each and to download it

programaty, 2024-01-16:
I just haven't found a way to do it, if there it is please let me know

MartinPacker, 2024-01-16:
I would also like this - if it isn't already available.

aivarannamaa, 2024-02-25:
You should be able to Ctrl-click several items (Command-click on macOS) and then upload/download them all at once.

Also, selecting first item and then Shift-selecting last item should also select all items in between. 

In Windows there's a possible point of confusion -- after selecting several items, the last clicked item gets different background from the other selected items. Just don't pay attention to this.