thonny/thonny #3076: Thonny unable to start Python process due to FileNotFoundError

krivers, 2024-01-18:
Hi there,

One of my students is having trouble getting Thonny to work. When he opens the application it immediately generates the following error message and crashes the Python interpreter:

Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_backend/cp_launcher.py", line 69, in <module>
    MainCPythonBackend(target_cwd, options).mainloop()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_backend/cp_back.py", line 126, in __init__
    self._check_warn_avoided_sys_path_conflicts()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_backend/cp_back.py", line 183, in _check_warn_avoided_sys_path_conflicts
    self._check_warn_sys_path_conflict(name.split(".")[0])
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/cpython_backend/cp_back.py", line 188, in _check_warn_sys_path_conflict
    user_dir = os.getcwd()
FileNotFoundError: [Errno 2] No such file or directory

Process ended with exit code 1.

We tried reinstalling Thonny and switching to a different version of Python, but neither worked. Do you have any recommendations for how to fix this?

Thank you!
- Kelly

aivarannamaa, 2024-02-01:
Thank you for the report and sorry for the late reply! 

I tried to reproduce this situation, but was not able to.

If the problem still persists, then please select "Tools => Open Thonny data folder" and send me the configuration.ini from it.  This may give a clue.

aivarannamaa, 2024-09-10:
I just experienced this error in macOS Catalina, but only for two runs. Could not reproduce it more, so I still don't know what can cause this.