thonny/thonny #3080: Pico W WebREPL TypeError

scruss, 2024-01-21:
Thonny 4.1.4 (Ubuntu 22.04.3 LTS, Python 3.10.12, Tk 8.6.12)

Configured Webrepl using `import webrepl_setup` prompted process, then on trying to connect using Thonny, the following failure message:
```
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/dist-packages/thonny/plugins/rp2040/rp2040_back.py", line 28, in <module>
    launch_bare_metal_backend(RP2040Backend)
  File "/usr/local/lib/python3.10/dist-packages/thonny/plugins/micropython/bare_metal_backend.py", line 1738, in launch_bare_metal_backend
    connection = WebReplConnection(args["url"], args["password"])
  File "/usr/local/lib/python3.10/dist-packages/thonny/plugins/micropython/webrepl_connection.py", line 39, in __init__
    raise res
  File "/usr/local/lib/python3.10/dist-packages/thonny/plugins/micropython/webrepl_connection.py", line 64, in _ws_main
    await self._ws_connect()
  File "/usr/local/lib/python3.10/dist-packages/thonny/plugins/micropython/webrepl_connection.py", line 77, in _ws_connect
    self._ws = await websockets.connect(self._url, ping_interval=None)
  File "/usr/lib/python3/dist-packages/websockets/legacy/client.py", line 622, in __await_impl__
    transport, protocol = await self._create_connection()
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1103, in create_connection
    transport, protocol = await self._create_connection_transport(
  File "/usr/lib/python3.10/asyncio/base_events.py", line 1121, in _create_connection_transport
    protocol = protocol_factory()
  File "/usr/lib/python3/dist-packages/websockets/legacy/client.py", line 160, in __init__
    super().__init__(**kwargs)
  File "/usr/lib/python3/dist-packages/websockets/legacy/protocol.py", line 154, in __init__
    self._drain_lock = asyncio.Lock(
  File "/usr/lib/python3.10/asyncio/locks.py", line 78, in __init__
    super().__init__(loop=loop)
  File "/usr/lib/python3.10/asyncio/mixins.py", line 17, in __init__
    raise TypeError(
TypeError: As of 3.10, the *loop* parameter was removed from Lock() since it is no longer necessary

Process ended with exit code 1.
```

aivarannamaa, 2024-01-22:
Thank you for the report!

Can you please check, which version of `websockets` package do you have (Tools => Manage packages)?

I found https://github.com/python-websockets/websockets/issues/916 but according to comments, the problem should be fixed in version 11, which comes bundled with Thonny 4.1.4

scruss, 2024-01-22:
Thanks! Somehow it was picking up python3-websockets v9.1 from the system Python interpreter. Upgrading from pypi to v.12.

It took a few tries to connect via WebREPL, but i can access files now

closed 2024-01-25 by aivarannamaa