thonny/thonny #3083: Unable to flash ESP8266 micropython to ESP8266 WeMos D1 (identical to issue #2961)

Flipje1955, 2024-01-22:
flashing using the Thonny configurator and ESPTOOL works smoothly but after the hard reset, the Micropython prompt does not appear.  The successful suggestions per issue #2961 of flashmode = dio and slah_size = detect dont solve my issue. Tried to boards, same result. Using "keep" or "qio" did not help.

Studied the RST pin of the board with a scope but it reurns to a "clean" HIGH after the process, Had a similar issue when setting up a ESP32-CAM but there playing with dts / rts settings resolved the issue. The logfiles do not give me any guidance. The configuration file shows:
**[ESP8266]**
port = auto
interrupt_on_connect = True
sync_time = True
local_rtc = True
restart_interpreter_before_run = True

I am using Thonny 4.1.4 on a W10 system. 

Would think there are quite some of these boards successfully set-up, so I must be doing something wrong. Welcome any guidance on what to change / add. Thank you.

Flipje1955, 2024-01-24:
First of all, thank you for your input. Tried exactly per your suggestions but to no avail: Result is just "Device is busy or does not respond. Your options: etc". 
If it is relevant: that text comes some 3 seconds after first printing a a screen-wide line.

Thonny selects V1.21.1 as MP version but using V1.21.0 or even an very recentely generated V1.23 does not change the outcome. I can see the onboard led flashing in sync with the upload indicator on the screen, which confirms there is a handshake with the device, something that further is confirmed when I play with the "configuration options" like "Query Device Info":

COM6 flash_id
esptool.py v4.6.2
Serial port COM6
Connecting....
Detecting chip type... Unsupported detection protocol, switching and trying again...
Connecting....
Detecting chip type... ESP8266
Chip is ESP8266EX
Features: WiFi
Crystal is 26MHz
MAC: 44:17:93:1f:71:d4
Uploading stub...
Running stub...
Stub running...
Manufacturer: 5e
Device: 4016
Detected flash size: 4MB
Hard resetting via RTS pin...
Done!

Thank you again for you support !

ADBeasley, 2024-01-24:
Checking on the main machine my only other thoughts are:

My version of Thonny is 4.1.3 on a M1 iMac connected via USB-C to Micro USB in one cable (no adapters)
My board version is v1.0.0:
Detecting chip type... ESP8266
Chip is **ESP8285N08**

So we have slightly different boards (I think mine is a cheap clone) but tis the only one I have handy at the mo - sorry.

Flipje1955, 2024-01-24:
thank you for your input and assistance !

Tried it on Rasp Pi 5 with the latest Thonny release;  same result.

Have installed ESPTOOL on my PC and will try (again) if that route solves the issue.

ADBeasley, 2024-01-24:
Be interesting to see if you can flash it with C code using the Arduino IDE or not - I found I could every time but getting MP on was a real pain (not as much as learning C/C++ though!).

Sorry I cannot help more - good look.