thonny/thonny #3084: installed Micropython on ESP-WROOM-02D not work

DanSkareda, 2024-01-23:
Hello. I have an ESP-WROOM-02D board (contains ESP8266ex). I am connected to it by a UART programmer. I can upload projects from the Arduino IDE and they work. So I started Thonny and wanted to use it to upload the Micropython firmware. The process went through, but the board in the shell window started sending endless "red nonsense". Everything is in the attachment. I am doing something wrong? The same procedure worked for me in the case of ESP32C3 (with micropython firmware for ESP32C3), the board answered "readable". I keep trying this over and over but every time I flash the micropython firmware for ESP8266 into ESP8266Ex, it does the same thing.
[image]
[image]
[image]

DanSkareda, 2024-01-23:
Solved in another forum with advice to change the flash mode.
[image]
[image]

DanSkareda, 2024-01-23:
I tried all possible flash modes in turn and they all have an unpleasant feature - after waking up ESP from deepsleep, a few lines of "red nonsense" are displayed. The ESP32C3 did not do this. So I don't know if the error is in Thonny's code or in Micropython's code.
[image]

aivarannamaa, 2024-02-01:
Sorry for the late reply.

I'm not too familiar with this, but I've read, that ESP-s use different baud rate at initial phase of their start-up, hence the garbage when read with 115200. Perhaps the same happens when waking from the deep sleep. Unfortunately I don't have any ideas how Thonny could detect the need for different baud rate (and to be honest, I'd rather not make the logic even more complex, even if I knew a solution).