thonny/thonny #3094: Batch script to launch Thonny's shell

G-Boronad, 2024-01-29:
Hye, 
Can you help me to write au batch script to launch Thonny's shell . 
I'd like to use this script as a shortcut for jupyter lab.
Thanks
Gwendal

aivarannamaa, 2024-01-31:
I'm afraid I don't understand. Do you want a way to launch Terminal with the same envrionment variables defined and instructions displayed as if it was launched via Thonny's "Tools => Open system shell"? In this case you could investigate https://github.com/thonny/thonny/blob/bc3efeae747aeada6e280d23cc3825d7eea83cc2/thonny/plugins/system_shell/__init__.py#L13

G-Boronad, 2024-02-03:
Yes, that's exactly what I want to do.
I looked at your file... but it didn't help me.
Thanks