thonny/thonny #3096: Bundled Python 3.10.11.0 has known vulnerabilities

nvpreport, 2024-01-31:
Microsoft MDE is reporting that the version of Python that is bundled with Thonny 4.1.4 has known weaknesses.

CVE-2007-4559
CVE-2023-27043
CVE-2023-24329

Could the version be upgraded to Python 3.10.13

Matt

aivarannamaa, 2024-01-31:
I might be worried if I ran this Python version on a server. Do you see a way these vulnerabilities could be exploited in the context where Thonny is used?

Toasty66, 2024-02-01:
I came here to mention the same problem.  From what I understand, 3.10.x is now in security patches only, so binary versions/installers are no longer being offered.  The developers should consider moving to a main support version such as 3.11.x or 3.12.x. where executables are released.

Toasty66, 2024-02-06:
> I might be worried if I ran this Python version on a server. Do you see a way these vulnerabilities could be exploited in the context where Thonny is used?

Sure. If it's running in a non-admin context (user space) and the vulnerability allowed the user to escape it and run code as an Administrator.  One of the issues concerns a directory traversal vulnerability where files can be overwritten.

aivarannamaa, 2024-02-10:
Thonny 5.0 will be bundled with latest Python 3.12

closed 2024-02-10 by aivarannamaa

nvpreport, 2024-02-12:
Fantastic 👍