thonny/thonny #3097: Waveshare RP2040-Zero support

raybellis, 2024-01-31:
I don't see a board profile for this board:

https://www.waveshare.com/rp2040-zero.htm

aivarannamaa, 2024-01-31:
What do you mean by board profile? Where do you expect to see it?

5ucur, 2025-07-07:
My guess as for where they would have expected to have seen it - as well as where I am not seeing it, which was what prompted me to look for issues on this topic - would be in the Options, subwindow "Install or update MicroPython", dropdown "variant".

It has listed e.g. Adafruit • Feather 2040, Adafruit • ItsyBitsy 2040, Pimoroni • Badger 2040 (various versions), WeAct • Studio RP2040, and many more variants of the 2040 which are not the [Waveshare RP2040-Zero](https://www.waveshare.com/wiki/RP2040-Zero). The page linked supplies its own UF2, but I have not found success in using it.

If this is the problem the issue opener has, it is upstream (the MicroPython site already does not offer Waveshare downloads), so, not an issue in Thonny. I have been using the Adafruit ItsyBitsy 2040's UF2 instead, with mixed results, but mostly successful.

Please let me know if another variant already in the dropdown is more or even fully compatible with this model.