thonny/thonny #3101: Typos in Ukrainian translation of Windows launcher.

Updaty, 2024-02-03:
Hello. There are some misprints in Ukrainian localisation of Thonny editor.

[image]
Here should be "Інструменти"("Tools" in English) instead of "Інстументи".

[image]
Here should be "Розкоментувати"("Uncomment" in English) instead of "Разкоментувати".

[image]
Here should be "Властивості"("Properties" in English) instead of "Власитвості".

[image]
And here should be "Перемістити до Кошика"/"Перемістити до Смітника"  instead of "Перемістити до Мусора". (The word "Кошик" which means "Basket", is used in Windows as translation for "Bin". If you want a word that literally means a bin, you can use "Смітник")

I also noticed that in the Ukrainian localization, some inscriptions, such as "New file" or "Save all files", are written in English.

Thank you for reading and supporting Ukraine!

aivarannamaa, 2024-02-07:
Thanks for noticing this! Please go to https://poeditor.com/join/project/Gh188fdYH6 and apply your fixes there. Before the next release I will download translation files from there.

thorn0, 2024-02-24:
It says "This project can't accept new contributors at this time. Please try again later" when I try to join the project on POEditor to fix these and other issues in the Ukrainian translation.

aivarannamaa, 2024-02-25:
@thorn0, please try once again!

thorn0, 2024-02-25:
Same result

aivarannamaa, 2024-02-25:
Thanks for the information! I sent POEditor an inquiry.