thonny/thonny #3103: Possible timing error when listing volumes with dbus-next

aivarannamaa, 2024-02-05:
I was double-tapping the reset button of Adafruit Trinket M0 while in the Uf2FlashingDialog, in order to put the device into bootloader mode, and received following exception:

```
Traceback (most recent call last):
  File "/usr/lib/python3.11/tkinter/__init__.py", line 1948, in __call__
    return self.func(*args)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/tkinter/__init__.py", line 861, in callit
    func(*args)
  File "/home/aivar/python_stuff/thonny/thonny/workdlg.py", line 216, in _keep_updating_ui
    self.update_ui()
  File "/home/aivar/python_stuff/thonny/thonny/plugins/micropython/base_flashing_dialog.py", line 149, in update_ui
    targets = self.find_targets()
              ^^^^^^^^^^^^^^^^^^^
  File "/home/aivar/python_stuff/thonny/thonny/plugins/micropython/uf2dialog.py", line 33, in find_targets
    for vol in list_volumes(skip_letters=["A"])
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aivar/python_stuff/thonny/thonny/misc_utils.py", line 112, in list_volumes
    return list_volumes_sync()
           ^^^^^^^^^^^^^^^^^^^
  File "/home/aivar/python_stuff/thonny/thonny/udisks.py", line 19, in list_volumes_sync
    return loop.run_until_complete(list_volumes())
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/home/aivar/python_stuff/thonny/thonny/udisks.py", line 66, in list_volumes
    connection_bus = await interface.get_connection_bus()
                     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/aivar/python_stuff/thonny/venv/lib/python3.11/site-packages/dbus_next/aio/proxy_object.py", line 118, in property_getter
    BaseProxyInterface._check_method_return(msg, 'v')
  File "/home/aivar/python_stuff/thonny/venv/lib/python3.11/site-packages/dbus_next/proxy_object.py", line 62, in _check_method_return
    raise DBusError._from_message(msg)
dbus_next.errors.DBusError: No such interface “org.freedesktop.DBus.Properties” on object at path /org/freedesktop/UDisks2/drives/Adafruit_Trinket_M0_970E2A2451314E5020312E380C1810FF
```

The dialog is polling for list of volumes 5 times per second. @jwillikers, can it be that the procedure listed the drives to process, but once it got to processing the drive corresponding to the device, the drive was not present anymore? What do you think of adding a try-exect somewhere, so that the rest of the drives can still be returned?

jwillikers, 2024-02-05:
Yeah, that sounds about right. I wonder if it would be easiest to just have a try-except that just continues with the next loop?

aivarannamaa, 2024-02-05:
Sounds good! Perhaps also log the error in the except-block?

commit 029ac3b9 (2024-02-12): Add better handling for udisks (#3108)

Add try-except blocks to catch potential DBus errors in list_volumes.
The try-except blocks log the error encountered and continue on.
This avoids potential issues when checking for volumes on USB devices which may be unplugged while processing.

Fixes #3103.
closed 2024-02-12 by aivarannamaa