thonny/thonny #3104: "Tools > Manage packages > Install from local file" crashes Thonny

jfeser, 2024-02-05:
Thonny version: 4.1.4
macOS: 13.6
Python: 3.10.11
Tk: 8.6.13

Attached a log showing the crash on macOS. This crash is also reproducible on Linux. I'm happy to get another log if that would help.

[thonny.log](https://github.com/thonny/thonny/files/14170657/thonny.log)

ironsightdesign, 2024-06-10:
Is this broken until v5? I dont see anywhere to download v5 betas.

dpwalker16, 2024-08-13:
I have the same problem with Thinny on Mac is there a work around?
Thonny version: 4.1.4
macOS: 14.6.1 (arm64)
Python: 3.10.11
Tk 8.6.13

abc1111111-sudo, 2024-10-28:
4.1.6 still didn't fix the issue for me even though the changelog says it should.

afreeland, 2025-01-29:
I still see this issue :(

aivarannamaa, 2025-05-31:
Fixed in 0060404261b67fca479b0b420dd9762411777ffd