thonny/thonny #3106: Problem while handling "Debugger response"

mathias-ted, 2024-02-09:
Hi, I am receiving this error while debbugging my code.

This is the error:

Traceback (most recent call last):
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\site-packages\thonny\workbench.py", line 1788, in event_generate
    handler(event)
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 1217, in _handle_debugger_progress
    _current_debugger.handle_debugger_progress(msg)
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 278, in handle_debugger_progress
    self._main_frame_visualizer.update_this_and_next_frames(msg)
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 425, in update_this_and_next_frames
    self._next_frame_visualizer.update_this_and_next_frames(msg)
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 411, in update_this_and_next_frames
    self._update_this_frame(msg, frame_info)
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 1018, in _update_this_frame
    DialogVisualizer._update_this_frame(self, msg, frame_info)
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 972, in _update_this_frame
    FrameVisualizer._update_this_frame(self, msg, frame_info)
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 470, in _update_this_frame
    self._expression_box.update_expression(msg, frame_info)
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 662, in update_expression
    self._load_expression(
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\debugger.py", line 756, in _load_expression
    for node in ast.walk(main_node):
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\ast.py", line 390, in walk
    todo.extend(iter_child_nodes(node))
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\ast.py", line 272, in iter_child_nodes
    for name, field in iter_fields(node):
  File "C:\Users\user\AppData\Local\Programs\Thonny\lib\ast.py", line 260, in iter_fields
    for field in node._fields:
AttributeError: 'NoneType' object has no attribute '_fields'

This is the code:

class BankAccount:
    def __init__(self, account_holder, balance=0):
        self.account_holder = account_holder
        self.balance = balance

    def deposit(self, amount):
        self.balance += amount
        print(f"Deposited ${amount}. New balance: ${self.balance}")

    def withdraw(self, amount):
        if amount <= self.balance:
            self.balance -= amount
            print(f"Withdrew ${amount}. New balance: ${self.balance}")
        else:
            print("Insufficient funds!")

    def get_balance(self):
        return self.balance

# Example usage:
if __name__ == "__main__":
    # Creating two bank accounts
    account1 = BankAccount("Alice", 1000)
    account2 = BankAccount("Bob")

    # Depositing and withdrawing from accounts
    account1.deposit(500)
    account2.deposit(1000)

    account1.withdraw(200)
    account2.withdraw(1500)  # This should print "Insufficient funds!"

    # Displaying final balances
    print(f"{account1.account_holder}'s final balance: ${account1.get_balance()}")
    print(f"{account2.account_holder}'s final balance: ${account2.get_balance()}")

The error occurs in this line of the code  '" print(f"Deposited ${amount}. New balance: ${self.balance}") ' ".

Current Operating system: Windows 10 pro 22H2 build 19045.3996
Python version: 3.12.0
Thoony version: 4.1.4

aivarannamaa, 2024-02-10:
Thank you for the report!

I was able to reproduce this problem with Python 3.12, but not with Python 3.10.

I'll investigate what has changed in 3.12 and what to do about it.

dukora53, 2024-02-10:
Hello. Last night I reinstalled Thonny to the latest version 4.1.4. I downloaded the latest stable OTA firmware for both ESP32 and 8266 from the Espressif site and started uploading to the 8266 board. After finishing programming, Shell -ThonnyIDE- began typing out various characters at high speed and flashing LEDs at high frequency, and Thonny "hardened". I had to open a new window - in the terminal with esptool.py erase everything. This happened with both ESP32 and 8266. I work in Debian 12, Python is 3.12, but both Debian and Python were with the previous version of Thonny. I'm not an IT, just a playing retiree who used to work with electronics. Thank you for your advice.

aivarannamaa, 2024-05-30:
I can't reproduce this with Python 3.12.3 anymore. I assume it was a problem in Python 3.12.0 and is fixed by now.

closed 2024-05-30 by aivarannamaa