thonny/thonny #3109: cannot write to circuit or repeat script run on CPX on macOS

soraraso42, 2024-02-10:
### what I have done 

- uninstall and reinstall Thonny
- update/re-install in bootloader mode (UF2)  in Thonny’s ‘configurate intepreter panel”
- auto-detect port
- select port 
- mannualy drag 8.2.9 driver to CPX drive in macOS file

<br>### expected behavior
- runs script every time
- can write to CPX via Thonny

<br>### current behavior
- runs once ( with exit then cannot connect to device. Back End not found error.
- saves correctly once then program freezes up. Infinite ‘saving’ prompt window.

<br>### steps to reproduce bug
- runs a script or save to board a second time

<br>### possible solutions
- change interpreter to local python
- stop process
- quit Thonny eject CPX
- start again

<br>### context 
- I am running python script on CPX from Thonny but only the first run can run. Subsequent runs on the exact same script will first show ‘no backend’, after going through the bootloader process once more, the backend is displayed in the lower right corner but script still cannot run because the ‘file cannot save’ error.

<br>### my environment

macOS 13.2.1 (22D68)
Thonny 4.1.4
Python 3.10.11
Tk 8.6.13
CPX driver Adafruit CircuitPlayground 8.2.9
mini-USB-to-typeC-cable by Amazon Basics

<br>### Error message

#### while trying to save file
```
Traceback (most recent call last):
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/workbench.py", line 2027, in toolbar_handler
    handler(*args)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/running.py", line 527, in cmd_run_current_script
    self.execute_current("Run")
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/running.py", line 459, in execute_current
    filename = editor.save_file()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/editors.py", line 312, in save_file
    result = self.write_remote_file(save_filename, content_bytes, save_copy)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/editors.py", line 398, in write_remote_file
    if "error" in result:
TypeError: argument of type 'NoneType' is not iterable

```

#### while trying to run a script

```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'Run' (ConnectionError: read failed: [Errno 6] Device not configured).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your CircuitPython device and try again.

Process ended with exit code 1.
Couldn't find the device automatically. 
Check the connection (making sure the device is not in bootloader mode) or choose
"Configure interpreter" in the interpreter menu (bottom-right corner of the window)
to select specific port or another interpreter.

Process ended with exit code None.

```

#### error message after update/re-install uf2 for CPX 8.2.9 and run script

```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'write_file' (ConnectionError: read failed: [Errno 6] Device not configured).
See Thonny's backend.log for more info.
You may need to press "Stop/Restart" or hard-reset your CircuitPython device and try again.

Process ended with exit code 1.
```

aivarannamaa, 2024-02-25:
Thank you for the report and sorry for the late reply!

Can you post the script you are trying to run? 

Also, after reproducing this problem, please select "Tools => Open Thonny data folder" and send me the backend.log. This may give necessary information for debugging this problem.

soraraso42, 2024-02-25:
Hi, thanks for getting back to me.

**the script** I was trying to run
```
# Write your code here :-)
from adafruit_circuitplayground import cp
import time

# Define Morse code mappings for letters and numbers
morse_code = {
    'A': '.-', 'B': '-...', 'C': '-.-.', 'D': '-..', 'E': '.', 'F': '..-.', 'G': '--.',
    'H': '....', 'I': '..', 'J': '.---', 'K': '-.-', 'L': '.-..', 'M': '--', 'N': '-.',
    'O': '---', 'P': '.--.', 'Q': '--.-', 'R': '.-.', 'S': '...', 'T': '-', 'U': '..-',
    'V': '...-', 'W': '.--', 'X': '-..-', 'Y': '-.--', 'Z': '--..',
    '0': '-----', '1': '.----', '2': '..---', '3': '...--', '4': '....-', '5': '.....',
    '6': '-....', '7': '--...', '8': '---..', '9': '----.',
    ' ': ' '
}

cp.pixels.brightness = 0.6
def display_morse_code(message):
    for char in message.upper():
        if char == ' ':
            time.sleep(3)  # Pause between words
        else:
            code = morse_code.get(char)
            for symbol in code:
                if symbol == '.':
                    # cp.red_led = True
                    cp.pixels.fill((144, 50, 109))
                    time.sleep(0.2)  # Dot duration
                    cp.pixels.fill((0, 0, 0))
                elif symbol == '-':
                    # cp.red_led = True
                    cp.pixels.fill((144, 50, 109))
                    time.sleep(0.5)  # Dash duration
                    cp.pixels.fill((0, 0, 0))
                # cp.red_led = False
                time.sleep(0.2)  # Pause between symbols
            time.sleep(0.4)  # Pause between characters

message = "HAPPY BIRTHDAY SHO HIRANO"
display_morse_code(message)

```
**the problem:**
first run successful
after removing one comment and trying to run again, program stuck in infinite saving to circuit mode.

**the backend log** 

>10:47:28.264 [MainThread] INFO    thonny: Thonny version: 4.1.4
10:47:28.264 [MainThread] INFO    thonny: cwd: /Users/a42
10:47:28.264 [MainThread] INFO    thonny: original argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python', '-s', '-u', '-B', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/circuitpython/cirpy_back.py', "{'clean': True, 'port': '/dev/cu.usbmodem101', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'CircuitPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
10:47:28.264 [MainThread] INFO    thonny: sys.executable: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10
10:47:28.264 [MainThread] INFO    thonny: sys.argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/circuitpython/cirpy_back.py', "{'clean': True, 'port': '/dev/cu.usbmodem101', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'CircuitPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
10:47:28.264 [MainThread] INFO    thonny: sys.path: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/circuitpython', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python310.zip', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/lib-dynload', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages']
10:47:28.264 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=1, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
10:47:28.264 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': '/dev/cu.usbmodem101', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'CircuitPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
10:47:28.339 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
10:47:28.339 [MainThread] INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type CircuitPythonBackend
10:47:28.339 [MainThread] INFO    thonny: TIME/MODS 0.100 before prepare (+73 modules)
10:47:28.339 [MainThread] INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors [... cut ...]
    filename = editor.save_file()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/editors.py", line 312, in save_file
    result = self.write_remote_file(save_filename, content_bytes, save_copy)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/editors.py", line 398, in write_remote_file
    if "error" in result:
TypeError: argument of type 'NoneType' is not iterable

soraraso42, 2024-02-25:
Hi thanks for getting back to me.

**The script** 
```
# Write your code here :-)
from adafruit_circuitplayground import cp
import time

# Define Morse code mappings for letters and numbers
morse_code = {
    'A': '.-', 'B': '-...', 'C': '-.-.', 'D': '-..', 'E': '.', 'F': '..-.', 'G': '--.',
    'H': '....', 'I': '..', 'J': '.---', 'K': '-.-', 'L': '.-..', 'M': '--', 'N': '-.',
    'O': '---', 'P': '.--.', 'Q': '--.-', 'R': '.-.', 'S': '...', 'T': '-', 'U': '..-',
    'V': '...-', 'W': '.--', 'X': '-..-', 'Y': '-.--', 'Z': '--..',
    '0': '-----', '1': '.----', '2': '..---', '3': '...--', '4': '....-', '5': '.....',
    '6': '-....', '7': '--...', '8': '---..', '9': '----.',
    ' ': ' '
}

cp.pixels.brightness = 0.6
def display_morse_code(message):
    for char in message.upper():
        if char == ' ':
            time.sleep(3)  # Pause between words
        else:
            code = morse_code.get(char)
            for symbol in code:
                if symbol == '.':
                    # cp.red_led = True
                    cp.pixels.fill((144, 50, 109))
                    time.sleep(0.2)  # Dot duration
                    cp.pixels.fill((0, 0, 0))
                elif symbol == '-':
                    # cp.red_led = True
                    cp.pixels.fill((144, 50, 109))
                    time.sleep(0.5)  # Dash duration
                    cp.pixels.fill((0, 0, 0))
                # cp.red_led = False
                time.sleep(0.2)  # Pause between symbols
            time.sleep(0.4)  # Pause between characters

message = "HAPPY BIRTHDAY SHO HIRANO"
display_morse_code(message)

```

**The problem**
script first run no issue
after removing a line of comment and trying to run again
Thonny stuck and froze

**The backend log** 
>10:47:28.264 [MainThread] INFO    thonny: Thonny version: 4.1.4
10:47:28.264 [MainThread] INFO    thonny: cwd: /Users/myusrname
10:47:28.264 [MainThread] INFO    thonny: original argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/Resources/Python.app/Contents/MacOS/Python', '-s', '-u', '-B', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/circuitpython/cirpy_back.py', "{'clean': True, 'port': '/dev/cu.usbmodem101', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'CircuitPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
10:47:28.264 [MainThread] INFO    thonny: sys.executable: /Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/bin/python3.10
10:47:28.264 [MainThread] INFO    thonny: sys.argv: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/circuitpython/cirpy_back.py', "{'clean': True, 'port': '/dev/cu.usbmodem101', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'CircuitPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}"]
10:47:28.264 [MainThread] INFO    thonny: sys.path: ['/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/plugins/circuitpython', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python310.zip', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/lib-dynload', '/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages']
10:47:28.264 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=1, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
10:47:28.264 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Starting backend, args: {'clean': True, 'port': '/dev/cu.usbmodem101', 'dtr': None, 'rts': None, 'submit_mode': None, 'interrupt_on_connect': True, 'write_block_size': None, 'write_block_delay': None, 'proxy_class': 'CircuitPythonProxy', 'sync_time': True, 'validate_time': False, 'local_rtc': True}
10:47:28.339 [MainThread] INFO    thonny.plugins.micropython.bare_metal_backend: Initial submit_mode: raw_paste, write_block_size: 127, write_block_delay: 0, read_block_size: 0
10:47:28.339 [MainThread] INFO    thonny.plugins.micropython.mp_back: Initializing MicroPythonBackend of type CircuitPythonBackend
10:47:28.339 [MainThread] INFO    thonny: TIME/MODS 0.100 before prepare (+73 modules)
10:47:28.339 [MainThread] INFO    thonny: NEW MODS ['__future__', '_bz2', '_compression', '_datetime', '_heapq', '_lzma', '_posixsubprocess', '_queue', '_struct', '_tkinter', 'array', 'binascii', 'bz2', 'datetime', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'gettext', 'heapq', 'lzma', 'math', 'ntpath', 'pathlib', 'platform', 'queue', 'select', 'selectors', 'serial', 'serial.serialposix', 'serial.serialutil', 'shlex', 'shutil', 'signal', 'struct', 'subprocess', 'termios', 'textwrap', 'thonny.backend', 'thonny.base_file_browser', 'thonny.codeview', 'thonny.config_ui', 'thonny.editors', 'thonny.languages', 'thonny.m [... cut ...]
    filename = editor.save_file()
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/editors.py", line 312, in save_file
    result = self.write_remote_file(save_filename, content_bytes, save_copy)
  File "/Applications/Thonny.app/Contents/Frameworks/Python.framework/Versions/3.10/lib/python3.10/site-packages/thonny/editors.py", line 398, in write_remote_file
    if "error" in result:
TypeError: argument of type 'NoneType' is not iterable