thonny/thonny #3111: Download problem in 'Manage Packages...'

Electro4z, 2024-02-12:
Hi,
Using Thonny 4.1.4 on Windows 10 Enterprise.
When using "Tools" -> "Manage Packages...", and trying to install 'micropython-modbus', I get the following error code:

install --progress-bar off --user micropython-modbus
Collecting micropython-modbus
  Using cached micropython-modbus-2.3.7.tar.gz (26 kB)
ERROR: micropython-modbus from https://files.pythonhosted.org/packages/a4/00/ac6cd7ec350915f14cb0caf8fe590f181b629dcd227adc8682678e5803a8/micropython-modbus-2.3.7.tar.gz does not appear to be a Python project: neither 'setup.py' nor 'pyproject.toml' found.
Process returned with code 1

Downloaded it manually and unzipped the file, but where are the libs stored?

aivarannamaa, 2024-02-25:
Thanks for the report!

Are you sure you are using MicroPython back-end? What does it say in the lower-right corner of Thonny's main window when you select "Tools" -> "Manage Packages..." ?

Electro4z, 2024-02-25:
MicroPython (Raspberry Pi Pico)...
I downloaded the package manually and go on from there since I didn't get it to work by using "Tools" -> "Manage Packages..."...

closed 2024-02-25 by Electro4z