thonny/thonny #3113: OpenSSL 1.1.1.x in 4.1.4 is now EOL

Toasty66, 2024-02-12:
Just wanted to mention that the OpenSSL included in 4.1.4 is EOL (End of Life) as of September 2023. As you are working on the 5.0 release, please consider moving to the latest supported version of OpenSSL e.g., 3.2.x.

aivarannamaa, 2024-02-13:
For Windows and macOS I'll be using whatever is provided in the official Python build. For Linux I'll stop publishing the binary bundle altogether -- Thonny 5.0 will come as a script which installs system dependencies, creates a virtual environment and installs `thonny` together with its Python dependencies.

Toasty66, 2024-02-22:
What is the projected release ETA for Thonny 5.0?  Our information security teams are advising that we remove Thonny 4.x because of the EOL/Vulnerable Python and OpenSSL versions.

aivarannamaa, 2024-02-25:
> What is the projected release ETA for Thonny 5.0

Maybe a month, maybe 3 months, maybe even more. Hard to predict.

> Our information security teams are advising that we remove Thonny 4.x because of the EOL/Vulnerable Python and OpenSSL versions.

Out of curiosity, what kind of scenario your information security team is afraid of?

Toasty66, 2024-02-26:
You have end-of-life versions of python that aren't patched with the latest security releases, and OpenSSL version is the same i.e., no more patches against security threats.  On a machine where Thonny is run in a non-admin user content, one of the current python vulnerabilities could allow the user to escape the non-admin environment and delete files. 

Said another way, Thonny release cadence should include updates to these end-of-life products and vulnerabilities.  If you are going to bundle Python, it's on you to keep in current and especially when there are known security issues.  If 5.0, or a release of 4.x that is patched, is three months or more away, best cyber practice would be to stop using/deploying it.

Toasty66, 2024-02-26:
I get it, but that's not how cyber security works.  Malicious tool kits will search and try to exploit as many things that they know about as possible, and it's not about what is or is not the highest plausibility.  This is about good sound project management, where including end of life/known vulnerable pieces of software should be addressed in a timely manner and shouldn't be a debate.  

My expectation would be that the Thonny project would be updated before bundled products go end-of-life, and certainly to release patched versions of the release when bundled products with vulnerabilities are discovered.

closed 2024-05-30 by aivarannamaa