thonny/thonny #3118: Searching for a circuitpython package returns no results under Manage Packages for Circuitpython Device.

sctv, 2024-02-20:
When searching for circuitpython packages such as DHT library `Adafruit_CircuitPython_DHT` (https://github.com/adafruit/Adafruit_CircuitPython_DHT) does not appear.
The micro-python version does `dht @ micropython-lib` - DHT11 & DHT22 temperature/humidity sensor driver.
If Adafruit_CircuitPython_DHT is put in a requirements.txt file, the dialog does correctly install it.
There seems to be an issue with searching the circuitpython libraries specifically.

aivarannamaa, 2024-02-25:
Thank you for the report!

Does searching "adafruit-circuitpython-dht" (hyphens instead of underscores) work for you?

sctv, 2024-02-25:
Yes, that does work.  However, searching for dht only finds the micropython dht11 and pure python dht versions. So there seems to be a difference with how the circuitpython libraries are indexed.