thonny/thonny #3119: NameError: __file__  and __main__ are not defined

Donatussss, 2024-02-21:
Hi, I'm getting this issue:
```python
print(f'main: {__main__}')
```
```repl
>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 3, in <module>
NameError: name '__main__' isn't defined
>>>
```
```python
print(f'file: {__file__}')
```
```repl
>>> %Run -c $EDITOR_CONTENT

MPY: soft reboot
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name '__file__' isn't defined
>>>
```
I'm using
- Thonny 4.1.4
- Windows 10 Pro
- Version 22H2
- Installed on 15/‎04/‎2021
- OS build 19045.2965

Any help regarding the same will be appreciated

_Originally posted by @Donatussss in https://github.com/thonny/thonny/issues/1014#issuecomment-1952848374_

aivarannamaa, 2024-02-25:
When you run MicroPython code in Thonny with F5 (or Run => Run current script), Thonny sends the code from current editor to MicroPython REPL for execution. That's why `__file__` is not defined in this case -- for MicroPython there is no file involved. That's also how CPython's REPL and `-c` command line switch work.

In MicroPython `__file__` is defined only if you save your script on the device as "main.py" and let MicroPython run it by rebooting the device.

`__name__`, on the other hand, should be defined also in the REPL. Which MicroPython version are you using? In MicroPython 1.22.1 for Raspberry Pi Pico, this variable is defined.

Donatussss, 2024-02-26:
Hi @aivarannamaa, thank you for replying and for your explanation.
I'm using MicroPython v1.21.0. Yes, `__name__` is defined. However, it is not useful in a script where I wish to save project details in a csv using the file name as the searchable item in the csv whenever the file is executed from Thonny.
It seems it is not possible to implement such a feature given your explanation
Thank you again

closed 2024-02-26 by Donatussss