thonny/thonny #3121: issue creating new directory on micropython device (unsupported operand type(s) for +: 'NoneType' and 'str')

echo-lalia, 2024-02-23:
Here's the error:

```
Traceback (most recent call last):
  File "C:\Users\Ethan\AppData\Local\Programs\Thonny\lib\tkinter\__init__.py", line 1921, in __call__
    return self.func(*args)
  File "C:\Users\Ethan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 885, in mkdir
    self.perform_mkdir(parent, name.strip())
  File "C:\Users\Ethan\AppData\Local\Programs\Thonny\lib\site-packages\thonny\base_file_browser.py", line 1262, in perform_mkdir
    path = (parent_dir + self.get_dir_separator() + name).replace("//", "/")
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
```

Error is popping up for me when trying to create a directory on the root of my esp32s3 device with MicroPython. I have done this before so I'm not certain what's triggering this issue now, but it's happening repeatedly. It seems to coincide with the files not showing up right away on the MicroPython device, but that might just be a coincidence.

aivarannamaa, 2024-02-25:
Thank you for the report!

>  It seems to coincide with the files not showing up right away on the MicroPython device, but that might just be a coincidence.

Does it mean you are trying to create the directory while the file browser is empty (and there actually are files or folders on the device)?

echo-lalia, 2024-02-25:
I always refresh it until the I can see the files, but sometimes it takes a few tries to get them to load, and if that happens, the path above the files usually does not load (on the MicroPython device only)