thonny/thonny #3123: Ukraine flag in software is a No Goal by the FOSS CoC

webdeb, 2024-02-23:
If you don't remove the Ukraine flag, you just don't respect the Free and Open Source Code of Conduct.
https://hub.fosstodon.org/coc/

Would you be fine when your phone would suddenly start to tell you a political message, when you look at it?
This is a no-go. Just keep your political views out of software, it sucks!

aivarannamaa, 2024-02-25:
This kind of arrogant trolling (including #3122, https://github.com/thonny/thonny/issues/2967#issuecomment-1961827183 and https://github.com/thonny/thonny/issues/3010#issuecomment-1961823804) won't help you, especially in the world of open source software.

closed 2024-02-25 by aivarannamaa

webdeb, 2024-02-25:
You are an arrogant troll, you abuse your repository superpower to provide your own political opinions, which do not belong into software per-se.