thonny/thonny #3125: Can't install packages

Sylvain-BROCAS, 2024-02-26:
Hi,
I have an error with my Thonny IDE. When I want to use the package manager, it can find the packages I'm looking for, but it can't install it, it displays this error : 

```
install micropython-upip
Error 'type' object is not subscriptable
```
I t[image]

ried Thonny 4.1.4 and 4.0.2, I tried to install several packages, I tried to install different versions of a package. Nothningto do, I have to manually install the packages.

I'm on PC Windows 10
Thank you

aivarannamaa, 2024-03-17:
Thank you for the report!

With which Thonny version did you create this screenshot? 

I tried searching for `micropython-upip` in Thonny 4.1.4 and could find and install it without problems onto Raspberry Pi Pico.

Sylvain-BROCAS, 2024-03-22:
Hi,
I took the screenshot with the 4.0.2 version. I will try again as soon as I can but I tried several different modules and none of them successed to install with always the same error (micropython-stepper, upip,

aivarannamaa, 2025-05-31:
Please reopen if the issue persists in the latest version