thonny/thonny #3126: Secure Coding Warning MacOS Sonoma 14.3.1

vsquared, 2024-02-27:
Congratulations on creating a very nice Python editor for beginners.  I really enjoy using it and has a been a big help in starting to learn Python programming.  On MacOS Sonoma 14.3.1 I get the following warning which should be easy to fix in the source code:

2024-02-27 16:32:35.917 Python[15428:696646] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.

aivarannamaa, 2024-03-01:
Thank you for kind words!

This warning is fixed in the latest macOS Python 3.12 build, which will be part of Thonny 5.0.

vsquared, 2024-03-17:
You probably already know it, but Thonny also runs PyObjc like a champ.  I
just discovered that last night and already have a runnable objc shell in
Thonny:

# start code

from Cocoa import (NSObject, NSApplication, NSApp, NSWindow, NSButton,
NSSound, NSMakeRect,
NSBezelStyleCircular, NSBezelStyleRounded, NSBeep, NSMenu, NSMenuItem,
NSTitledWindowMask, NSClosableWindowMask,
NSResizableWindowMask, NSMiniaturizableWindowMask,NSBackingStoreBuffered)
from PyObjCTools import AppHelper

global _wndH
global _wndW

class AppDelegate(NSObject):

    def buildMenu(self):
        menubar = NSMenu.alloc().init()
        app_menu_item = NSMenuItem.alloc().init()
        menubar.addItem_(app_menu_item)
        NSApp.setMainMenu_(menubar)
        app_menu = NSMenu.alloc().init()
        quit_menu_item =
(NSMenuItem.alloc().initWithTitle_action_keyEquivalent_("Quit",
"terminate:", "q"))
        app_menu.addItem_(quit_menu_item)
        app_menu_item.setSubmenu_(app_menu)

    def buildWindow(self):
        _wndH = 400
        _wndW = 400
        #  wnd.initWithContentRect_styleMask_backing_defer_(frame, 15, 2, 0)
        frame = (NSMakeRect(0, 0, _wndH, _wndW))
        wndStyle = NSTitledWindowMask + NSClosableWindowMask +
NSMiniaturizableWindowMask + NSResizableWindowMask
        wnd =
NSWindow.alloc().initWithContentRect_styleMask_backing_defer_(frame,
wndStyle, NSBackingStoreBuffered, False)
        wnd.setTitle_("HelloWorld")
        wnd.center()
        wnd.orderFrontRegardless()

        myBtn = NSButton.alloc().initWithFrame_(NSMakeRect(10.0, _wndH -
50, 90, 30))
        wnd.contentView().addSubview_(myBtn)
        myBtn.setBezelStyle_(NSBezelStyleRounded)
        myBtn.setTitle_("Hello!")
        myBtn.setAction_("myBtnAction:")

        quitBtn = NSButton.alloc().initWithFrame_(NSMakeRect(_wndW - 60,
10, 50, 30))
        wnd.contentView().addSubview_(quitBtn)
        quitBtn.setBezelStyle_(NSBezelStyleCircular)
        quitBtn.setTitle_("Q")
        quitBtn.setAction_("terminate:")

    def applicationDidFinishLaunching_(self, aNotification):
       self.buildMenu()
       self.buildWindow()

    def myBtnAction_(self, sender):
        print("Hello World!")
        NSBeep()

def main():
    NSApplication.sharedApplication()
    delegate = AppDelegate.alloc().init()
    NSApp().setDelegate_(delegate)
    AppHelper.runEventLoop()

main()

#end code

closed 2024-05-30 by aivarannamaa