thonny/thonny #3131: Waiting for python version 3.12

amaxcz, 2024-03-11:
Уважаемые авторы этого замечательного инструмента для обучения детей, пожалуйста:

Сделайте версию для python 3.12. Крайне востребовано.

+ так же хочется иметь возможность устанавливать шрифт для окон типа Variables.
+ впишите в рекомендации юзерам использование шрифтов Fira Code (который умеет лигатуры) для кода 
+ и MesloLGS NF для IO font (умеет все powerlevel фишки итп)

aivarannamaa, 2024-03-17:
The next version of Thonny will come bundled with Python 3.12.

The ability to change font size in Variables view is already in the TODO list (#3055).

Regarding specific fonts -- the users can choose whatever fonts they like (or whatever someone recommends them to use), so I don't see the need to do something about this.

christophprokop, 2024-03-22:
> The next version of Thonny will come bundled with Python 3.12.

That's great! Thanks! Is there a public road map or a planned release date for this?

aivarannamaa, 2024-03-24:
> Is there a public road map or a planned release date for this?

https://github.com/thonny/thonny/milestones/5.0.0 is the TODO list for stable release. Most likely I'll postpone some items from this list.

I can't promise any release date.

codeBodger, 2025-07-29:
Is there any sense as to when this will be released?  Could a mini-version be made, almost identical to 4.1.7, but with 3.12?