thonny/thonny #3134: error in installing dlib eventhough I have an updated python and I already install cmake.

mntnnc3058, 2024-03-12:
[image]
It is showing this kind of error when Im installing dlib. I already installed the cmake and I have updated python verision.

aivarannamaa, 2024-03-17:
dlib seems to be written in C++. Are you sure you have all the required components in place for building the library from source?

mntnnc3058, 2024-03-17:
Hello @aivarannamaa! I think yes I already have all required components. But just incase, may I know all required components for me to check, thank you!

aivarannamaa, 2024-03-24:
I've never used dlib. I guess you should ask in a dlib forum. I'm willing to investigate, if you are able to install dlib outside of Thonny but not with Thonny's package manager.

closed 2024-03-24 by aivarannamaa