thonny/thonny #3135: TODO pane picks up erroneous comments (Scunthorpe problem)

athompson673, 2024-03-12:
I'm trying to refer to an `OSError` in a comment without it becoming a todo list item. I think the regex should be changed to prevent the list of match words (TODO, BUG, FIXME, etc) from being found as a substring in a larger word. I'm pretty sure this could be done by adding `\b` around the capture group to require word boundaries before and after the matched word.

https://github.com/thonny/thonny/blob/f21193ec0f1e7ca8e1f5c70a4b22c7b59fb9f274/thonny/plugins/todo_view.py#L106

<pre><code>r_ex = r"^.*((#.*[\t ]*<b>\b</b>(TODO|BUG|FIXME|ERROR|NOTE|REMARK)<b>\b</b>([:\t ]*))(.*))$" </code></pre>

athompson673, 2024-03-14:
In a more significant change, I would personally require the keyword to directly follow the `#` symbol with only optional whitespace before it, however I can recognize that is simply a personal preference.

`r_ex = r"^.*((#\s*(TODO|BUG|FIXME|ERROR|NOTE|REMARK)\b([:\t ]*))(.*))$"`

aivarannamaa, 2024-03-17:
Thank you for the report!

I agree that the regex should be more strict. Your regex will be in next Thonny version.

closed 2024-03-17 by aivarannamaa