thonny/thonny #3137: pyglet window doesn't close properly

Friendly-Aid, 2024-03-13:
I just started using pyglet, and I found that when I tried to close the window it wouldn't close, and the window stopped responding. I tried to run the same code on pythons IDLE and it closed properly. I don't know if this helps at all but I do get `2024-03-13 13:15:09.365 Python[1807:59998] ApplePersistenceIgnoreState: Existing state will not be touched. New state will be written to /var/folders/8s/szj3l83j1fl79lyp1v1xhmg40000gn/T/org.python.python.savedState` when assigning the window.

thank you for taking your time to read this.

aivarannamaa, 2024-03-17:
Thank you for the report!

Which Pyglet version are you using? I was not able to reproduce this problem with version 2.0.14.

Friendly-Aid, 2024-03-17:
I am using 2.0.14. here is a video of the error if that helps.

https://github.com/thonny/thonny/assets/163163308/03ee296b-94fd-4141-be85-0b765f25c581

Friendly-Aid, 2024-03-17:
I am using 2.0.14. here is a video showing my error incase that helps.

https://github.com/thonny/thonny/assets/163163308/0905b4c4-6f59-41d2-b2d8-da9205821f51

aivarannamaa, 2024-03-24:
I'm using macOS, but I tried the code you sent me earlier, which had `on_close` event handler. The program in your screencast does not have it. What happens if you add it?

Friendly-Aid, 2024-03-24:
I used my example code in the first comment and got the loading lollipop when I hover over the pyglet screen. What version of python are you using? Because the default for thonny is 3.10 unless installed via pip. I changed that to python3.12 which I installed from 'python.org'. Also the autocomplete doesn't seem to work when using the python.org interpreter and I am likely going to change to a pip installed version of thonny. Also the bug or error with the pyglet screen happens on the pip version as well for me.

aivarannamaa, 2024-03-29:
I was using Python 3.12