thonny/thonny #3139: WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.

DanHuntlyCollege, 2024-03-15:
Still have this issue with Thonny 4.1.4 
Mac OS Sonoma
Updated to Python 13.12.1 but I think waiting for Thonny V5.0?
How is that coming along? Other issue logs pointed to a Jan / Feb 2024 release, but they have been closed.
Thanks, Dan

aivarannamaa, 2024-03-17:
According to my tests, this warning doesn't show up in Python 3.12.2

DanHuntlyCollege, 2024-03-17:
Thanks for replying Annamaa! So if I update python, Thonny uses that in the background, and will fix the issue...? I very much appreciate your quick response! Wonderful work, and this great app for my students.Best regards, Dan

DanHuntlyCollege, 2024-03-18:
Hi there Aivar (sorry got you name wrong... )
I have had a look at how I can upgrade Python, outside of Thonny, and gave
this a try.
I also had a look at all the Thonny versions to see if any are bundled with
python 3.12.2, but they are all older versions and come with just python
3.10.
Is there a way to make Thonny use the version of Python, or is there a beta
of thonny with the new version bundled?

Thanks for your help, Dan Howard

# Gui Learning Program
# import the Gui code library we want to use
from tkinter import *

# this code creates a first window. The 'mainloop' part stops the program
from ending

# Function for the Quit button
def quit():
    main_window.destroy()

def hello():
    print('Hello')

#function to use the main window to add a button
def main():
    main_window = Tk()
    Button(main_window, text='Quit',command=quit) .grid()
    Button(main_window, text='Enter Data', command=hello) .grid()
    main_window.mainloop()

# run the main window function
main()

aivarannamaa, 2024-03-24:
> Is there a way to make Thonny use the version of Python, or is there a beta of thonny with the new version bundled?

The beta is still in works, but you can install Pytho 3.12 separately and make Thonny use it (Tools => Options => Interpreter).

Please note that the warning you are talking about is just a warning. It is totally OK to simply ignore it for now.

closed 2024-03-24 by aivarannamaa