thonny/thonny #3146: command prompt opens

mrpal5a, 2024-03-23:
Every time I open Thonny It will open the command prompt and even if I try to write something keyboard doesn't write

https://github.com/thonny/thonny/assets/107930441/eece4bcc-8795-4958-aa26-cbc1857d3efd

Please help me in solving this problem. 
Above is the screen recording, so that you can understand my problem clearly.
Thank you

aivarannamaa, 2024-03-24:
When googling the error code `3221225876` (and its hex form 0xc0000194) I found `STATUS POSSIBLE DEADLOCK`, but I have no idea what could be causing this.

Do I understand correctly, that you don't need to run any code for this problem to happen?

mrpal5a, 2024-03-24:
So what I supposed to do. Stop using Thonny? or wait