thonny/thonny #3149: Watch-changes / Sync mode

dustinlacewell, 2024-03-26:
It'd be pretty handy if Thonny grew a watch mode, keeping the remote in sync with the project. The use-case is when Thonny is not being utilized as an editor, but rather for its reliable management of the MP device.

aivarannamaa, 2024-03-29:
Have you tried using existing command line tools and libraries for managing the MP device? If not, then you should see at least [mpremote](https://pypi.org/project/mpremote/) and [pyboard.py](https://github.com/micropython/micropython/blob/master/tools/pyboard.py)

dustinlacewell, 2024-03-30:
I recommended to mpremote they add watch functionality, and they recommended I checkout rshell. They don't have it either, so I opened an issue there to recommend the feature also. 

Everybody should add watch functionality! Hehe, it's a multiplier to quick iterations during early hot development and debugging.

Anyway, just a suggestion! Thanks for replying.

closed 2024-07-12 by aivarannamaa