thonny/thonny #3150: Unable to input Chinese

idea-hk, 2024-03-30:
IM inputbox dosent appear when texting , hence is unable to input chinese. Other programs however work well with fcitx5, such as terminal and geany. 

The common way to trigger input method is using ctrl + space, and has been altered to some other combination, to avoid onflict with the auto-complete shortcut of thonny.

I can only managed to input chinese via copy and paste, which is inconvenient doing so. 

Raspberry Pi 4
RapsiOS bullseye 64bit
LXDE
Fcitx5
Python 3.9.2
Tk 8.6.11
Thonny 4.0.1

[image]
[image]
[image]

idea-hk, 2024-03-30:
The imput method Rime works well with thonny running Win11
[image]

bh1rio, 2024-12-22:
At present, Thonny is just short of the function of inputting Chinese to become a perfect Python IDE.

bh1rio, 2024-12-29:
https://core.tcl-lang.org/tk/tktview/2902214fff

@aivarannamaa 
is it the error ?

aivarannamaa, 2025-01-15:
Thonny 5.0 will bundle a newer version of Tcl/Tk. Hopefully this will fix this. Please re-open if it does not.

closed 2025-01-15 by aivarannamaa

bh1rio, 2025-01-21:
Excuse me, when is the version 5.0 planned to be released?

aivarannamaa, 2025-01-21:
Hopefully before the spring.