thonny/thonny #3153: Interrupt working program on connect unchecked doesn't seem to work.

dcmcshan, 2024-04-03:
[image]

I have unchecked "Interrupt working program on connect"

I have confirmed this in configuration.ini

[ESP32]
port = /dev/cu.usbserial-210
interrupt_on_connect = False
submit_mode = raw_paste
sync_time = True
local_rtc = True
restart_interpreter_before_run = False
webrepl_url = ws://192.168.4.1:8266/
webrepl_password = AVSMCT
last_configurations = [{'run.backend_name': 'ESP32', 'ESP32.port': 'webrepl', 'ESP32.webrepl_url': 'ws://192.168.4.1:8266/'}]

But it seems that connecting still results in soft reset:

MPY: soft reboot

I have confirmed that SerialTools can connect/disconnect without soft reset.

aivarannamaa, 2024-04-30:
You could try manually adding 

* either `dtr = True` and `rts = False` 
* or `dtr = False` and `rts = False` 

to the `[ESP32]` block in the configuration file. Make sure Thonny is closed while you edit the file, otherwise your edits will be lost.

dcmcshan, 2024-11-09:
I think the problem fundamentally is that I am using USB CDC interface and so I don't have the ability to use drt/rts to trigger reset.  Does Thonny send any ctrl signal on connect?

aivarannamaa, 2025-01-15:
Thonny 4 uses `pyserial`-s default connection parameters and I'm not 100% sure, what they are. Thonny 5 will allow tweaking DTR and RTS from the configuration dialog.