thonny/thonny #3157: Thonny uses local time instead of UTC in micropython

wohltat, 2024-04-08:
In micropython there is no difference between time.localtime and time.gmtime.
https://github.com/orgs/micropython/discussions/9545#discussioncomment-3822001
That's why it make sense that ntptime sets the time in UTC.

As far as i know, Thonny sets the time on the micropython board automatically on connect.
But the time is inconsistent with ntptime. It should also be UTC and not time.localtime of the host.
This can causes different behaviour depending on wether you reboot your board connected to Thonny or standalone.

I meantioned this behavior here already:
https://github.com/orgs/micropython/discussions/13217#discussioncomment-7891438

aivarannamaa, 2024-04-30:
Have you tried unchecking "Use local time in real time clock" under MicroPython options (Tools => Options => Interpreter => MicroPython ...) ?

wohltat, 2024-05-01:
Thanks, that seems to do it. 
UTC is now shown on `time.localtime()` 
Sorry, i missed the option.

closed 2024-05-01 by wohltat