thonny/thonny #3158: ValueError: AST node line range is not valid

wohltat, 2024-04-09:
Trying to run any of the Pygame Zero examples fails with
```PROBLEM IN THONNY'S BACK-END: Exception while handling 'Run' (ValueError: AST node line range (1000000, 1) is not valid).
See Thonny's backend.log for more info.

Process ended with exit code 1.
```

The backend log seems to contain nothing obvious:
```
13:41:14.266 [MainThread] INFO    thonny: Thonny version: 4.1.4
13:41:14.267 [MainThread] INFO    thonny: cwd: /home/username/.config/Thonny/leave_this_empty
13:41:14.267 [MainThread] INFO    thonny: original argv: ['/home/username/Documents/programming_test/pygamezero/pgzero/examples/flappybird/env/bin/python3', '-u', '-B', '/usr/lib/python3.11/site-packages/thonny/plugins/cpython_backend/cp_launcher.py', '/home/username/Documents/programming_test/pygamezero/pgzero/examples/basic', "{'run.warn_module_shadowing': True}"]
13:41:14.267 [MainThread] INFO    thonny: sys.executable: /home/username/Documents/programming_test/pygamezero/pgzero/examples/flappybird/env/bin/python3
13:41:14.267 [MainThread] INFO    thonny: sys.argv: ['/usr/lib/python3.11/site-packages/thonny/plugins/cpython_backend/cp_launcher.py', '/home/username/Documents/programming_test/pygamezero/pgzero/examples/basic', "{'run.warn_module_shadowing': True}"]
13:41:14.267 [MainThread] INFO    thonny: sys.path: ['/usr/lib/python3.11/site-packages', '/usr/lib/python3.11/site-packages/thonny/plugins/cpython_backend', '/usr/lib/python311.zip', '/usr/lib/python3.11', '/usr/lib/python3.11/lib-dynload', '/home/username/Documents/programming_test/pygamezero/pgzero/examples/flappybird/env/lib/python3.11/site-packages']
13:41:14.267 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, safe_path=False, int_max_str_digits=-1)
13:41:14.267 [MainThread] INFO    thonny: TIME/MODS 0.015 Before constructing backend (+28 modules)
13:41:14.267 [MainThread] INFO    thonny: NEW MODS ['_heapq', '_locale', '_posixsubprocess', '_queue', 'encodings.ascii', 'errno', 'faulthandler', 'fcntl', 'fnmatch', 'heapq', 'importlib._abc', 'importlib.util', 'ipaddress', 'locale', 'math', 'ntpath', 'pathlib', 'queue', 'select', 'selectors', 'signal', 'subprocess', 'thonny.backend', 'thonny.plugins', 'thonny.plugins.cpython_backend', 'thonny.plugins.cpython_backend.cp_back', 'urllib', 'urllib.parse']
13:41:14.267 [MainThread] INFO    thonny: TIME/MODS 0.000 Before MainBackend 
13:41:14.268 [MainThread] INFO    thonny: TIME/MODS 0.000 After MainBackend 
13:41:14.278 [MainThread] INFO    thonny.backend: Loading plugins
13:41:14.278 [MainThread] INFO    thonny: TIME/MODS 0.011 Before loading plugins (+6 modules)
13:41:14.278 [MainThread] INFO    thonny: NEW MODS ['__future__', '_sysconfigdata__linux_x86_64-linux-gnu', 'pkgutil', 'platform', 'pydoc', 'sysconfig']
13:41:14.279 [MainThread] INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['/usr/bin', '/usr/lib/python311.zip', '/usr/lib/python3.11', '/usr/lib/python3.11/lib-dynload', '/home/username/.local/lib/python3.11/site-packages', '/usr/lib/python3.11/site-packages', '/home/username/.config/Thonny/plugins/lib/python3.11/site-packages']
13:41:14.280 [MainThread] INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.birdseye_backend'
13:41:14.280 [MainThread] INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.dock_user_windows_backend'
13:41:14.281 [MainThread] INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.flask_backend'
13:41:14.281 [MainThread] INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.matplotlib_backend'
13:41:14.281 [MainThread] INFO    thonny.backend: Loading plugin 'thonny.plugins.backend.pgzero_backend'
13:41:14.287 [MainThread] INFO    thonny: TIME/MODS 0.009 After loading plugins (+6 modules)
13:41:14.288 [MainThread] INFO    thonny: NEW MODS ['thonny.plugins.backend', 'thonny.plugins.backend.birdseye_backend', 'thonny.plugins.backend.dock_user_windows_backend', 'thonny.plugins.backend.flask_backend', 'thonny.plugins.backend.matplotlib_backend', 'thonny.plugins.backend.pgzero_backend']
13:41:14.292 [MainThread] INFO    thonny: TIME/MODS 0.005 Beginning of mainloop 
13:41:14.293 [MainThread] INFO    thonny: TIME/MODS 0.001 Sending message ToplevelResponse 
13:41:14.293 [MainThread] INFO    thonny.backend: Loading Jedi
13:41:14.293 [MainThread] INFO    thonny: TIME/MODS 0.000 Before loading Jedi 
13:41:14.294 [MainThread] INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['/usr/bin', '/usr/lib/python311.zip', '/usr/lib/python3.11', '/usr/lib/python3.11/lib-dynload', '/home/username/.local/lib/python3.11/site-packages', '/usr/lib/python3.11/site-packages', '/home/username/.config/Thonny/plugins/lib/python3.11/site-packages']
13:41:14.375 [MainThread] INFO    thonny: TIME/MODS 0.082 After loading Jedi (+134 modules)
13:41:14.376 [MainThread] INFO    thonny: NEW MODS ['_blake2', '_bz2', '_compat_pickle', '_compression', '_ctypes', '_hashlib', '_json', '_lzma', '_pickle', '_struct', 'binascii', 'bz2', 'colorama', 'colorama.ansi', 'colorama.ansitowin32', 'colorama.initialise', 'colorama.win32', 'colorama.winterm', 'ctypes', 'ctypes._endian', 'difflib', 'filecmp', 'gc', 'hashlib', 'jedi', 'jedi._compatibility', 'jedi.api', 'jedi.api.classes', 'jedi.api.completion', 'jedi.api.completion_cache', 'jedi.api.environment', 'jedi.api.errors', 'jedi.api.exceptions', 'jedi.api.file_name', 'jedi.api.helpers', 'jedi.api.interpreter', 'jedi.api.keywords', 'jedi.api.project', 'jedi.api.refactoring', 'jedi.api.refactoring.extract', 'jedi.api.strings', 'jedi.cache', 'jedi.common', 'jedi.debug', 'jedi.file_io', 'jedi.inference', 'jedi.inference.analysis', 'jedi.inference.arguments', 'jedi.inference.base_value', 'jedi.inference.cache', 'jedi.inference.compiled', 'jedi.inference.compiled.access', 'jedi.inference.compiled.getattr_static', 'jedi.inference.compiled.mixed', 'jedi.inference.compiled.subprocess', 'jedi.inference.compiled.subprocess.functions', 'jedi.inference.compiled.value', 'jedi.inference.context', 'jedi.inference.docstring_utils', 'jedi.inference.docstrings', 'jedi.inference.filters', 'jedi.inference.flow_analysis', 'jedi.inference.gradual', 'jedi.inference.gradual.annotation', 'jedi.inference.gradual.base', 'jedi.inference.gradual.conversion', 'jedi.inference.gradual.generics', 'jedi.inference.gradual.stub_value', 'jedi.inference.gradual.type_var', 'jedi.inference.gradual.typeshed', 'jedi.inference.gradual.typing', 'jedi.inference.gradual.utils', 'jedi.inference.helpers', 'jedi.inference.imports', 'jedi.inference.lazy_value', 'jedi.inference.names', 'jedi.inference.param', 'jedi.inference.parser_cache', 'jedi.inference.recursion', 'jedi.inference.references', 'jedi.inference.signature', 'jedi.inference.syntax_tree', 'jedi.inference.sys_path', 'jedi.inference.utils', 'jedi.inference.value', 'jedi.inference.value.decorator', 'jedi.inference.value.dynamic_arrays', 'jedi.inference.value.function', 'jedi.inference.value.instance', 'jedi.inference.value.iterable', 'jedi.inference.value.klass', 'jedi.inference.value.module', 'jedi.inference.value.namespace', 'jedi.parser_utils', 'jedi.plugins', 'jedi.plugins.django', 'jedi.plugins.flask', 'jedi.plugins.pytest', 'jedi.plugins.registry', 'jedi.plugins.stdlib', 'jedi.settings', 'json', 'json.decoder', 'json.encoder', 'json.scanner', 'lzma', 'parso', 'parso._compatibility', 'parso.cache', 'parso.file_io', 'parso.grammar', 'parso.normalizer', 'parso.parser', 'parso.pgen2', 'parso.pgen2.generator', 'parso.pgen2.grammar_parser', 'parso.python', 'parso.python.diff', 'parso.python.errors', 'parso.python.parser', 'parso.python.pep8', 'parso.python.prefix', 'parso.python.token', 'parso.python.tokenize', 'parso.python.tree', 'parso.tree', 'parso.utils', 'pickle', 'pydoc_data', 'pydoc_data.topics', 'shutil', 'struct', 'zipfile', 'zlib']
```

Before I created a new Interpreter with a new environment folder and installed pgzero.

-----------

Thonny 4.1.4
Python 3.11.8

wohltat, 2024-04-09:
This seems to happen only in 'Pygame Zero mode'.

If i add `import pgzrun` and `pgzrun.go()` some of them (`circle.py`) seems to run.
Others (`asteroids.py`) need more configuration and fail with other error messages.

All examples work outside of Thonny when i run them via `pgzrun`.

wohltat, 2024-04-09:
The error `NameError: name 'storage' is not defined` resulted from different pgzero versions.

pip:
pgzero.__version__ = '1.2.1'

github:
pgzero.__version__ = '1.3.dev0'

After installing pgzero outside of Thonny, i could use version '1.3.dev0' in Thonny. 

Now i can workaround the initial issue and create a runner script like `run_flappybird.py` and use it instead of `pgzrun`.
```
import pgzero.runner

pgzero.runner.load_and_run('flappybird.py')
```

wohltat, 2024-04-09:
It seems to be still broken in the current version Thonny 5.0.0b1-dev.

wohltat, 2024-04-09:
Adding `end_lineno` with the same value as `lineno` in `thonny/plugins/backend/pgzero_backend.py` gets rid of the error message.

```
go.lineno = 1000000
go.end_lineno = 1000000
```

I guess this works as it was intended in Thonny, i.e. adding `import pgzrun` at the beginning and `pgzrun.go()` at the end, like it is described in the docs (https://pygame-zero.readthedocs.io/en/stable/ide-mode.html).

The catch is that this is not enough as i described above.

aivarannamaa, 2024-05-01:
Thank you for this thorough report! It looks like recent Python versions have more strict requirements about AST nodes. I'll fix it later.

PR #3552 (2025-02-22): Fix pygame zero mode by making the AST node valid
Closes #3158

Thank you for this great project!

Would be really great if we could get this in. We use Thonny for CoderDojo workshops, and without this we have to tell kids to switch to a terminal and run the script manually from there.

Thanks!
closed 2025-02-22 by aivarannamaa