thonny/thonny #3159: Word wrap/ Wrap text

divyamokal, 2024-04-09:
Pls add Word wrap/ Wrap text option in Thonny. It is very much needed function in the software. Except this Thonny is a good IDE for practice

aivarannamaa, 2024-05-01:
I wouldn't mind making it optional (although I don't like the idea myself, see #906), but unfortunately word-wrapping conflicts with the way line numbers are implemented in Thonny.

I'll postpone this issue until more pressing issues are solved or someone else wants to solve it.

pfuller304, 2024-07-10:
@aivarannamaa "but unfortunately word-wrapping conflicts with the way line numbers are implemented in Thonny."
I really like Thonny, it's the best python IDE I've used (teaching in Secondary school)
Would it be worth refining the way line numbers are generated so switching on Word Wrap would leave a gap where the line numbers are, on the wrapped line and move the next line number one line below(I'm not explaining very well, sorry)
Cheers

aivarannamaa, 2025-01-22:
I would say just use hard wraps, so that you don't need soft wrapping ;)