thonny/thonny #3160: Cannot run Thonny from source, KeyError: 'bordercolor'

wohltat, 2024-04-09:
I cannot start Thonny from source in linux.

First i cloned Thonny (`git clone github.com/thonny/thonny`) then I followed the description in https://github.com/thonny/thonny/wiki/Running-from-the-source.
After starting i get the following error

```
Traceback (most recent call last):
  File "/home/username/build/thonny/thonny/main.py", line 66, in run
    bench = workbench.Workbench(parsed_args)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/username/build/thonny/thonny/workbench.py", line 194, in __init__
    self._init_containers()
  File "/home/username/build/thonny/thonny/workbench.py", line 824, in _init_containers
    "nw": AutomaticNotebook(
          ^^^^^^^^^^^^^^^^^^
  File "/home/username/build/thonny/thonny/ui_utils.py", line 608, in __init__
    super().__init__(master, closable=closable)
  File "/home/username/build/thonny/thonny/custom_notebook.py", line 88, in __init__
    super().__init__(master, background=self.style_conf["bordercolor"])
                                        ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'bordercolor'
```

aivarannamaa, 2024-05-01:
Thank you for the report!

Which Linux distro, which Python version and which Tk version are you using? Tk version should be shown when you run `wish`.

wohltat, 2024-05-01:
Updated the versions in the initial post.

Strangely if i run Thonny with `ipython3 -m thonny` it works but with `python3 -m thonny` it doesn't.
Python and tk version seem to be the same in both cases.
I used both commands just right after each other in the same venv.

I'll try if i can see some differences in the console logs.

wohltat, 2024-05-01:
It was using the 'ipython` not from the venv but external. 
If i use the ipython from the venv then i get the same error message:

```
> venv/bin/ipython3 -m thonny
12:28:11.367 [MainThread] INFO    thonny: Thonny version: 5.0.0b1-dev
12:28:11.367 [MainThread] INFO    thonny: cwd: /home/username/build/thonny
12:28:11.367 [MainThread] INFO    thonny: original argv: ['/home/username/build/thonny/venv/bin/python', 'venv/bin/ipython3', '-m', 'thonny']
12:28:11.368 [MainThread] INFO    thonny: sys.executable: /home/username/build/thonny/venv/bin/python
12:28:11.368 [MainThread] INFO    thonny: sys.argv: ['/home/username/build/thonny/thonny/__main__.py']
12:28:11.368 [MainThread] INFO    thonny: sys.path: ['/home/username/build/thonny/venv/bin', '/home/username/build/thonny/thonny/vendored_libs', '/usr/lib/python311.zip', '/usr/lib/python3.11', '/usr/lib/python3.11/lib-dynload', '', '/home/username/build/thonny/venv/lib/python3.11/site-packages']
12:28:11.368 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=0, no_user_site=0, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, safe_path=False, int_max_str_digits=-1)
12:28:11.411 [MainThread] INFO    thonny.workbench: Starting Workbench
12:28:11.514 [MainThread] INFO    thonny.workbench: First time adding 'ShellView' to []
12:28:11.514 [MainThread] INFO    thonny.workbench: First time adding 'AssistantView' to []
12:28:11.514 [MainThread] INFO    thonny.workbench: Creating runner
12:28:11.516 [MainThread] INFO    thonny.workbench: Start loading plugins
12:28:11.571 [MainThread] INFO    thonny.workbench: First time adding 'AstView' to ['ShellView']
12:28:11.572 [MainThread] INFO    thonny.workbench: First time adding 'StackView' to ['AssistantView']
12:28:11.572 [MainThread] INFO    thonny.workbench: First time adding 'ExceptionView' to ['AstView', 'ShellView']
12:28:11.573 [MainThread] INFO    thonny.plugins.event_logging: Starting logging user events into '/home/username/build/thonny/venv/.thonny/user_logs/2024-05-01_12-28-11__unknown.jsonl'
12:28:11.573 [MainThread] INFO    thonny.workbench: First time adding 'FilesView' to []
12:28:11.574 [MainThread] INFO    thonny.workbench: First time adding 'HeapView' to []
12:28:11.574 [MainThread] INFO    thonny.workbench: First time adding 'HelpView' to []
12:28:11.575 [MainThread] INFO    thonny.workbench: First time adding 'NotesView' to ['HelpView']
12:28:11.575 [MainThread] INFO    thonny.workbench: First time adding 'ObjectInspector' to ['StackView', 'AssistantView']
12:28:11.576 [MainThread] INFO    thonny.workbench: First time adding 'OutlineView' to ['NotesView', 'HelpView']
12:28:11.577 [MainThread] INFO    thonny.workbench: First time adding 'PackagesView' to ['AstView', 'ExceptionView', 'ShellView']
12:28:11.578 [MainThread] INFO    thonny.workbench: First time adding 'TodoView' to ['AstView', 'ExceptionView', 'PackagesView', 'ShellView']
12:28:11.578 [MainThread] INFO    thonny.workbench: First time adding 'VariablesView' to ['NotesView', 'OutlineView', 'HelpView']
12:28:11.579 [MainThread] INFO    thonny.workbench: Done loading plugins
12:28:11.605 [MainThread] WARNING thonny.workbench: Could not find UI theme 'clam', switching to default
12:28:11.643 [MainThread] ERROR   thonny.main: Internal launch or mainloop error
Traceback (most recent call last):
  File "/home/username/build/thonny/thonny/main.py", line 66, in run
    bench = workbench.Workbench(parsed_args)
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/username/build/thonny/thonny/workbench.py", line 194, in __init__
    self._init_containers()
  File "/home/username/build/thonny/thonny/workbench.py", line 824, in _init_containers
    "nw": AutomaticNotebook(
          ^^^^^^^^^^^^^^^^^^
  File "/home/username/build/thonny/thonny/ui_utils.py", line 608, in __init__
    super().__init__(master, closable=closable)
  File "/home/username/build/thonny/thonny/custom_notebook.py", line 88, in __init__
    super().__init__(master, background=self.style_conf["bordercolor"])
                                        ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
KeyError: 'bordercolor'
12:28:11.834 [MainThread] INFO    thonny.ui_utils: Parent y: 642, rooty: 642, vrooty: 0
12:28:11.835 [MainThread] INFO    thonny.ui_utils: Placing .!longtextdialog with geometry +237+822

```

The only substantial difference i can see is the following additional folders in `sys.path`:
```
'/home/username/.local/lib/python3.11/site-packages',
'/usr/lib/python3.11/site-packages'
```

The `WARNING thonny.workbench: Could not find UI theme 'clam', switching to default` also looks suspicious.

aivarannamaa, 2024-05-23:
I think I found the problem. The fix is commited to `master` branch.

wohltat, 2024-05-24:
Ok, the error is gone now. 
But still looks different. I does not seem to use my configuration like dark them, etc.

aivarannamaa, 2024-07-12:
Are you still experiencing problems with themes?

wohltat, 2024-07-12:
Seems good now. Either way i start it, the same config is used.

closed 2024-07-12 by wohltat