thonny/thonny #3161: can't install package in thonny ide for rpi pico w

Ransu242, 2024-04-10:
someong help me. im trying to install pyrebase for my rpi pico w using thonny ide i can't install pyrebase package it said it doesnt look like a micropython package and if i install it an error occur

image

Installing to temp directory
"C:\Program Files (x86)\Thonny\python.exe" -m thonny.plugins.micropython.minipip install --target C:\Users\USER\AppData\Local\Temp\tmpbm4c2jqz Pyrebase
Processing 'Pyrebase'
Querying package metadata from https://micropython.org/pi/Pyrebase/json
Could not find 'Pyrebase' from https://micropython.org/pi
Querying package metadata from https://pypi.org/pypi/Pyrebase/json
Inspecting version 3.0.27
'Pyrebase' will be installed with pip (not having single tar.gz asset).
Installing with pip: ['Pyrebase']
Collecting Pyrebase
Using cached Pyrebase-3.0.27-py3-none-any.whl.metadata (681 bytes)
Collecting gcloud==0.17.0 (from Pyrebase)
Using cached gcloud-0.17.0-py3-none-any.whl
Collecting oauth2client==3.0.0 (from Pyrebase)
Using cached oauth2client-3.0.0-py3-none-any.whl
Collecting pycryptodome==3.4.3 (from Pyrebase)
Using cached pycryptodome-3.4.3.tar.gz (6.5 MB)
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Collecting python-jwt==2.0.1 (from Pyrebase)
Using cached python_jwt-2.0.1-py2.py3-none-any.whl.metadata (7.5 kB)
Collecting requests==2.11.1 (from Pyrebase)
Downloading requests-2.11.1-py2.py3-none-any.whl.metadata (39 kB)
Collecting requests-toolbelt==0.7.0 (from Pyrebase)
Using cached requests_toolbelt-0.7.0-py2.py3-none-any.whl.metadata (11 kB)
Collecting httplib2>=0.9.1 (from gcloud==0.17.0->Pyrebase)
Using cached httplib2-0.22.0-py3-none-any.whl.metadata (2.6 kB)
Collecting googleapis-common-protos (from gcloud==0.17.0->Pyrebase)
Using cached googleapis_common_protos-1.63.0-py2.py3-none-any.whl.metadata (1.5 kB)
Collecting protobuf!=3.0.0.b2.post1,>=3.0.0b2 (from gcloud==0.17.0->Pyrebase)
Using cached protobuf-4.24.4-cp37-cp37m-win32.whl.metadata (540 bytes)
Collecting six (from gcloud==0.17.0->Pyrebase)
Using cached six-1.16.0-py2.py3-none-any.whl.metadata (1.8 kB)
Collecting pyasn1>=0.1.7 (from oauth2client==3.0.0->Pyrebase)
Using cached pyasn1-0.5.1-py2.py3-none-any.whl.metadata (8.6 kB)
Collecting pyasn1-modules>=0.0.5 (from oauth2client==3.0.0->Pyrebase)
Using cached pyasn1_modules-0.3.0-py2.py3-none-any.whl.metadata (3.6 kB)
Collecting rsa>=3.1.4 (from oauth2client==3.0.0->Pyrebase)
Using cached rsa-4.9-py3-none-any.whl.metadata (4.2 kB)
Collecting jws>=0.1.3 (from python-jwt==2.0.1->Pyrebase)
Using cached jws-0.1.3-py3-none-any.whl
Collecting pyparsing!=3.0.0,!=3.0.1,!=3.0.2,!=3.0.3,<4,>=2.4.2 (from httplib2>=0.9.1->gcloud==0.17.0->Pyrebase)
Using cached pyparsing-3.1.2-py3-none-any.whl.metadata (5.1 kB)
Using cached Pyrebase-3.0.27-py3-none-any.whl (9.6 kB)
Using cached python_jwt-2.0.1-py2.py3-none-any.whl (8.8 kB)
Downloading requests-2.11.1-py2.py3-none-any.whl (514 kB)
---------------------------------------- 514.8/514.8 kB 2.5 MB/s eta 0:00:00
Using cached requests_toolbelt-0.7.0-py2.py3-none-any.whl (52 kB)
Using cached httplib2-0.22.0-py3-none-any.whl (96 kB)
Using cached protobuf-4.24.4-cp37-cp37m-win32.whl (409 kB)
Using cached pyasn1-0.5.1-py2.py3-none-any.whl (84 kB)
Using cached pyasn1_modules-0.3.0-py2.py3-none-any.whl (181 kB)
Using cached rsa-4.9-py3-none-any.whl (34 kB)
Using cached six-1.16.0-py2.py3-none-any.whl (11 kB)
Using cached googleapis_common_protos-1.63.0-py2.py3-none-any.whl (229 kB)
Using cached pyparsing-3.1.2-py3-none-any.whl (103 kB)
Building wheels for collected packages: pycryptodome
Building wheel for pycryptodome (setup.py): started
Building wheel for pycryptodome (setup.py): finished with status 'error'
error: subprocess-exited-with-error

python setup.py bdist_wheel did not run successfully.
exit code: 1

[364 lines of output]
running bdist_wheel
[... 349 lines cut ...]
copying lib\Crypto\SelfTest\Signature\test_vectors\ECDSA\SigVer.rsp -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\ECDSA
creating build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5
copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5\SigGen15_186-2.txt -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5
copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5\SigGen15_186-3.txt -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5
copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5\SigVer15_186-3.rsp -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-v1.5
creating build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS
copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS\SigGenPSS_186-2.txt -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS
copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS\SigGenPSS_186-3.txt -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS
copying lib\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS\SigVerPSS_186-3.rsp -> build\lib.win32-3.7\Crypto\SelfTest\Signature\test_vectors\PKCS1-PSS
Skipping optional fixer: buffer
Skipping optional fixer: idioms
Skipping optional fixer: set_literal
Skipping optional fixer: ws_comma
running build_ext
[end of output]

note: This error originates from a subprocess, and is likely not a problem with pip.
ERROR: Failed building wheel for pycryptodome
Running setup.py clean for pycryptodome
Failed to build pycryptodome
ERROR: Could not build wheels for pycryptodome, which is required to install pyproject.toml-based projects

minipip returned with error code 1

image

aivarannamaa, 2024-05-01:
The logs tell me that you are trying to install `pyrebase` onto your MicroPython device. Is this what you wanted to do?

Ransu242, 2024-05-14:
Yes I can't install any libraries in my rpi pico w

aivarannamaa, 2024-07-12:
Sorry for the late reply!

`pyrebase` is not meant for MicroPython.

closed 2024-07-12 by aivarannamaa