thonny/thonny #3163: MUST Support  Palestine !

xkonglong, 2024-04-13:
I can not imagine what it is like to live in the Ukraine & Iraqi & Afghanistan & Palestine &Syria.

So, The software Need and MUST Support these countries !

Are the Palestinians not human ?

Or is the software author just hypocritically kind?

kompilainenn, 2024-04-13:
> I can not imagine what it is like to live in the Ukraine & Iraqi & Afghanistan & Palestine &Syria.
> 
> So, The software Need and MUST Support these countries !
> 
> Are the Palestinians not human ?

The author can have his own bugs in his head. 
If I understood right your humor, then you can just delete that Ukrainian icon by commenting some rows of code in workbench.py file:

#         self.add_command(
#             "SupportUkraine",
#             "help",
#             tr("Support Ukraine"),
#             self._support_ukraine,
#             image="Ukraine",
#             caption=tr("Support"),
#             include_in_toolbar=True,
#             group=101,
#         )

Another way : you can just fork the project and add or delete there all you don't like :-)

aivarannamaa, 2024-04-13:
I'll keep my focus on Ukraine.

closed 2024-04-13 by aivarannamaa

xkonglong, 2024-04-13:
it is not humor，Sure enough, Ukrainians and Palestinians are not equal in the eyes of some.

For all I know, Estonia may simply hate Russia more. Perhaps support for Ukraine is not due to a love of peace?

SergejGleithmann, 2024-04-16:
> it is not humor，Sure enough, Ukrainians and Palestinians are not equal in the eyes of some.
> 
> For all I know, Estonia may simply hate Russia more. Perhaps support for Ukraine is not due to a love of peace?

I do understand your analogy, but first singeling out a crime does not aquit others. The fact that the war in Ukraine is terrible does not imply other wars are unimportant. Otherwise one could not speak of anything because speaking of anything will always exclude and by this logic discriminate everything unmentioned. Second fact is that Estonia has a history with Russia and since the Developer of this project is from Estonia, he might have other priorities than you regarding the current security policy status. The developer of leaflet for example also has Ukraine support icons in his software because he himself is directly affected by the war as a Ukrainian citizen. He certainly has other concerns then the civil war in Yemen or the Israel-Palestine conflict. 
In the case of Estonia, it shares borders with Russia and has been surpressed and annexed during Sowjet times, so this is to be concidered. Singeling out certain wars is always arbitrary to some extend, but the author has explained his stance and you are free to disagree, however this does not mean you can impose anything onto this repository by opening redundant issues. If you think other wars should be discussed within Thonny follow the suggestion and fork this project. The masses will then decide which project they'll use. That is the open-source way.

xkonglong, 2024-04-17:
> I do understand your analogy, but first singeling out a crime does not aquit others.

It is not my purpose.

I don't think a tool should promote ideologies and political views.   

If it has to, it should at least have universal values or give the user a choice.

It shouldn't be imposed on others, it's neither democracy, nor freedom, nor human rights.

SergejGleithmann, 2024-04-17:
> I don't think a tool should promote ideologies and political views.

That in itself is an opinion that you are entitled to. However others might have other opinions on this.
 
> If it has to, it should at least have universal values or give the user a choice.

The choice here is, to fork this project and make your own variant of it.

> It shouldn't be imposed on others, it's neither democracy, nor freedom, nor human rights.

The use of this tool is not imposed by the maintainer. You are free to use any other alternative or to fork and change the code. You are the one imposing political neutrality on the maintainer of this project by this. He is free to code the way he wants in his repository and you are free to code in yours, as long as neither of you is violating laws or human rights. Adding a "support ukraine"-button is no crime and not against human rights therefore the maintainer is free to do so. You are free to disagree to adding this button and to make your own version but you cannot dictate, what the maintainer has to do in his repo in this case. This is the democratic principle here.

aivarannamaa, 2024-04-18:
@xkonglong, your questions have been answered already. We are disagreeing with each other, but it is not the end of the world. 

In the context of this project, you are not entitled to have your way - Thonny is a gift and it is your choice to either accept or reject it. You are welcome to make proposals, but I make changes only if I buy your arguments. In this case I don't and I've explained my reasons. Let's move on, please!

xkonglong, 2024-04-18:
是的,作为作者您当然有权利接受或拒绝建议. 

我自己也能通过修改代码实现工具的无属性. 

也是因为 MUST 让您产生了误会. 

做为开源社区的一员, 这不妨碍我对开源人员的尊敬.

正因为开源, 这个世界才会更加美好. 

我衷心的希望世界和平, 没有战争.   不论是乌克兰, 还是巴勒斯坦. 

由于这个项目具有了政治属性, 我才会在这个项目上讨论相关话题.  

当然, 结果令人失望. 

尽管我坚持认为:  给用户的选择权不是`用`或者`不用`.   

但我仍然尊重您的选择. 

为了避免翻译可能带来的歧义, 所以这段回复我选择了中文. 

若有不便,敬请谅解.

rosesonfire, 2024-12-22:
It's maybe a double standard because afaik most pro-Ukraine folks are pro-Israel and anti-Palestine.

kompilainenn, 2024-12-22:
> It's maybe a double standard because afaik most pro-Ukraine folks are pro-Israel and anti-Palestine.

If you compare any news about Russia-Ukrainian war and about Israel-Palestina war in west newspapers, you'll get the excellent examples that just will confirm your sentence