thonny/thonny #3169: Remote SSH suddenly crashes with this error

abhignan-rakshith, 2024-04-21:
Please help me: this is the error when trying to connect via ssh to pi@pi.local

Traceback (most recent call last):
  File "C:\Users\abhig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython_ssh\cps_back.py", line 213, in <module>
    backend = SshCPythonBackend(**args)
  File "C:\Users\abhig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython_ssh\cps_back.py", line 45, in __init__
    self._proc = self._start_main_backend()
  File "C:\Users\abhig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\plugins\cpython_ssh\cps_back.py", line 153, in _start_main_backend
    return self._create_remote_process(
  File "C:\Users\abhig\AppData\Local\Programs\Thonny\lib\site-packages\thonny\backend.py", line 820, in _create_remote_process
    raise RuntimeError(f"Got {ack!r} instead of expected {PROCESS_ACK!r}")
RuntimeError: Got 'File "/tmp/thonny-backend-4.1.4-pi/thonny/plugins/cpython_backend/cp_launcher.py", line 35' instead of expected 'OK'

Process ended with exit code 1.

Alle1307, 2024-04-21:
Same problem here. Normal ssh via terminal working normally. With fresh image after key reset.

abhignan-rakshith, 2024-04-21:
> Same problem here. Normal ssh via terminal working normally. With fresh image after key reset.

we need a fix/solution soon, working on final year project, also considering to switch to ubuntu os and ssh via PyCharm, i have professional edition