thonny/thonny #3175: Micropython output and webrepl

mdughera, 2024-04-27:
I have a program running on an ESP32 board that uses asyncio and periodically prints a message. 
Webrepl is configured and working correctly: I can connect to the board with the url http://myboard:8266 and see the output messages as soon as they have been generated. 
In webrepl I can also upload and download files with the program running. 

I configured Thonny interpreter to use ws://myboard:8266/ and see the message "Webrepl connected". 
I can confirm by the board leds that the program is still running, but I cannot see any message in Thonny and the board filesystem does not show up.
To see the board filesystem I need to stop the running program. 

The fact that I cannot see the program output is a bug of a feature not yet implemented?
Also, why do I need to stop the program to send/receive files?

TIA and kudos for what you've implemented.

aivarannamaa, 2024-07-12:
Sorry for the late reply!

Thonny is implemented so that it communicates with the device only via MicroPython REPL, therefore nothing can be done when a program is running.

To be honest -- I haven't considered WebREPL a priority, as communication with WebREPL is slow because of couple of old MicroPython bugs (https://github.com/micropython/micropython/issues/2497 and https://github.com/micropython/micropython/issues/6912). The slowness may be not apparent when using plain REPL, but Thonny exchanges quite a lot of data after connecting.

closed 2024-07-12 by aivarannamaa

mdughera, 2024-07-12:
Hi Aivar,

Thanks for the reply. For your information the use case I have in mind is not use webrepl as the main way for development, rather use Thonny also for the remote monitoring of the prints that my board produces when running and the deployment of new versions of the script.

I am currently using thonny and repl for the main development and then use the webrepl in browser to look what a "production" board is doing or send script updates. It would be nice to do also this "last mile" in thonny.

No matter what, many thanks for your work: at the moment is the tool that I prefer for Micropython.

Marco Dughera
HR & Sourcing Italy

Reply
Via Cardinal Massaia, 83
10147 - Torino - ITALY
phone: +39 011 29100
***@***.******@***.***>
www.reply.com<http://www.reply.com>

[https://images.reply.eu/img2/reply.png]

To be honest -- I haven't considered WebREPL a priority, as communication with WebREPL is slow because of couple of old MicroPython bugs (micropython/micropython#2497<https://github.com/micropython/micropython/issues/2497> and micropython/micropython#6912<https://github.com/micropython/micropython/issues/6912>). The slowness may be not apparent when using plain REPL, but Thonny exchanges quite a lot of data after connecting.