thonny/thonny #3180: Thonny entering "0" randomly when editing a file on RPi5

MoJeeza, 2024-05-01:
When editing a python file on Rpi5, Thonny inserts "0" at high speed (faster than I can delete by keeping the delete key pressed)..
I am using the Logitech Pebble 2 K380s kb (via bluetooth)
Attached file provides the OS on Rpi5 and Thonny info as well as the "0" being added when in input mode of  new file.
[thonny_kb_zeros.dox.docx](https://github.com/thonny/thonny/files/15174155/thonny_kb_zeros.dox.docx)

On RPi5 I have turned off the BlueTooth feature
Connecting to Rpi using RealVNC client/server on a Win11 laptop (latest version)
I don't get this issue using Thonny on Win11 laptop using same kb

Only happens on Rpi using only Thonny.

aivarannamaa, 2024-05-11:
Unfortunately I don't have any ideas what may cause this.

Does the same happen when you are using Python's official "IDE" called IDLE? I'm asking, because IDLE is using the same UI toolkit as Thonny.

Victor-Reginato, 2024-05-22:
I'm having the same problem, but the characters that are typed out of control are a bit random, sometimes it's 9, sometimes it's 0 and they only stop being typed when I start typing manually. I am also using Real VNC to connect remotely and I have a Raspberry Pi 4 B. This does not happen in any other application open on the Raspberry

MoJeeza, 2024-07-06:
> Unfortunately I don't have any ideas what may cause this.
> 
> Does the same happen when you are using Python's official "IDE" called IDLE? I'm asking, because IDLE is using the same UI toolkit as Thonny.

Sorry for my late reply..
Yes, it's happening in IDLE too. 

I start Thonny, Open a blank file and click the mouse in the window...zeros start getting typed at high speed, I run the code file that has nothng but "0"s and the console window opens up. I mouse-click in the console window and the "0"s start appearing there...I click on the "stop/restart backend" button and the "0"s in the console stop and that window is cleared with only ">>>".

I have connected Pico-W to the usb port on Rpi5 and using Thonny editing a file on Rpi5 or on Pico-W does exactly the same thing.

All software is the latest update.

upstairs-at-erics, 2024-07-26:
I am experiencing this as well - but my character being keyed is 4.
Using Real VNC to connect.
Closing and re-opening Thonny does not resolve the issue - A reboot does for a while, but it re-appears randomly.
I will look into the behavior when connected via no machine or rdp

UltimateCodeWarrior, 2024-08-12:
Had the same problem tonight.   Upgraded to Bookworm on Rpi 4B.   Installed VNC Viewer on Windows 11.   Connecting to Raspberry PI fine.   Thonny didn't launch at all after  the OS Came up for the first time, I had to reboot.  (That's another issue I posted).     So, I learned to my dismay that Picamera was completely revamped and busted, and now there was Picamera2, so I started slogging my way using Thonny to recode the broken interfaces that were shamelessly and irresponsibly orphaned. I mean the rest of the libraries worked fine, why did someone think it was a good idea to take a Flagship library and mothball it?   So I'm doing a test-fix-test cycle ... then out of nowhere I see these 999999999 being typed to infinity and beyond in my code window.  I close it down, relaunch Thonny ... a few seconds later, the Poltergeist starts typing the 9's again.  Got some sort of Ghost in the Machine.  It's only on Thonny, no other editor does this.      I did switch Thonny over to the normal mode or whatever it was called.   Not sure what the enhanced mode would do for me.

shiner501, 2024-08-30:
I have exactly the same issue.  Raspberry Pi headless with VNC enabled and connecting via Real VNC on MAC.  Random zeros.  Also have the issue on Visual Studio too so seems a VNC issue at the Pi end.   Will try initiating from windows as well.

Quite a critical problem for me.

upstairs-at-erics, 2024-08-30:
Interestingly it doesn’t happen if accessing using rdpSent from my iPad

UltimateCodeWarrior, 2024-08-30:
My Fix :   I left using Thonny.  I installed Pi-Apps, and switched over to Sublime Text and just running the python script from the command prompt.  Wasn't a bad solution, actually worked quite well considering what was happening in Thonny.    I'm wonder if some of this came about because I didn't do a clean install of the new 64 Bit OS on the SD Card, I think I might have upgraded the OS-in place with the old 32 bit OS still on the SD-card.   Pi-Apps let's you install a lot of applications quickly, but will eat up your SD Card's space quick!     PyCharm was too slow, it didn't recognize the python libraries out of the gate.        Pi Connect was  worked great as an alternate to using RealVNC, but didn't allow me to cut and paste from windows 11.   That was a deal breaker for me as RealVNC allowed smaller clipboards to be cut and pasted.     Then I took the big plunge.      I switched over to VSCode and started Running CoPilot, the PI OS Came to a crawl.    It seems as if the PI OS alone was eating up 25% of it's CPU, and stacking VSCode and Copilot was bringing it to it's knees. I am working on Computer Vision Project with the  PI Camera P and OpenCV and this just was not working.    Somewhere in the Mix Pi-Apps wanted me to do an update, the update hung in the middle, so I rebooted the PI (big mistake) and then I couldn't log back into the PI GUI!   Fortunately  I had a backup of my software code.    So I took the even bigger plunge,  switched to using VSCode and CoPilot on my Windows 11 PC.  Far Far superior in speed and usability, no typing lag, code autocomplete suggestions as well as code chat made creation of mundane python 3 functions effortless.    I setup Sanba Server to make a folder share between raspberry PI and networked that shared folder to My Windows 11 PC.   (Making that share connect from windows was tricky, but possible).     Now I put my code in that folder and that's where I load from in VSCode running on windows, that way the PI and Windows can share the same codebase.      I switched the PI to boot to the command prompt instead of the GUI.  With PI SSH Enabled,  I use Bitvise Tunnilier from Windows 11 to connect to the SSH on the Raspberry PI so I can run the python code from the command prompt as running from Windows won't work with the picam  and other linux libraries not recognized :)     The big win was that I could run the Raspberry PI without the OS alone eating up 25% of the cpu, let alone the IDE tools, and faster boots.  I want to make the python scripts startup at reboot anyhow, and so why boot to the GUI.   So I guess I took the long trip around the barn, but now I have leveled up with this setup and can make some rapid progress not having to fight the funkyness of the key-lean issues.  Big tapdance, but I'm better for it.

shiner501, 2024-08-31:
I also find it doesn’t occur when using RDP. However I then get unusable windows if I open a web browser and other applications. Others have also commented on this too.  My workaround for now so I can get on with my work until this is fixed is by using Raspberry Connect.

upstairs-at-erics, 2024-08-31:
Does anyone know how and where to raise an issue? It seems it's with VNC, which I think may be deprecated for free use.

shiner501, 2024-09-01:
> Interestingly it doesn’t happen if accessing using rdp

I get strange screen artefacts in many applications when using RDP.

aivarannamaa, 2024-10-12:
Please note that the alternative is to run Thonny on your development machine and use the SSH back-end (Run => Configure Interpreter => Remote Python 3)

JohnPool, 2024-11-07:
I have the same (RPI5 and Thonny4.1.4). At unexpected moments, the key you press exhibits the same behavior as when you hold it down: a flood of identical characters.