thonny/thonny #3183: Ubuntu versions 3.4.14 and 4.1.4 not working, but Windows vesion does

bikingfreak, 2024-05-02:
I had been using Thonny (I guess the 3.4.14) version for nearly a year and was happy with it. Some recent Ubuntu updates must have broken is as I now get binary output in the shell window. However, it works fine with the same ESP32 in Windows 10. 

Here is some typical output in Ubuntu 22.04 (it does not seem to work at all in Ubuntu 24.04): 
>> %Run -c $EDITOR_CONTENT
Error ntptime settime: -202
Program ESP32sensor V1.5.12 Started!, 2024/05/02  11:56:43
mac_serial_number: 237576828101656
�7�u���7��.�O"B�g��������D��'��._�&P������

aivarannamaa, 2024-05-11:
Does it change anything when you uncheck "Restart interpreter before running a script" in the interpreter options (Tools => Options => Interpreter)?

bikingfreak, 2024-05-12:
Aivar,
There is no check box for that in version 3.3.14. I have had sporadic
success when unplugging the USB and plugging it in again or changing to
another PC or to a Windows PC. Very often I cannot get the "Device is busy"
popup to disappear, no matter how often I click on "Stop/Restart Backend"
or soft restart or interrupt execution.

If I uncheck the box in version 4.0.1, using an ESP32 in Windows, I can
connect to the device. If I try the same thing with Ubuntu 22.04.04 with
the same ESP32, it does not connect, but ls -al /dev/ttyUSB* does not find
the device either. If I connect to a USB port in my USB hub, I find
/dev/ttyUSB0, but then Thonny produces the following output, which did not
happen in Windows 10.
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4728
load:0x40078000,len:14876
ho 0 tail 12 room 4
load:0x40080400,len:3368
entry 0x400805cc

Device is busy or does not respond. Your options:

  - wait until it completes current work;
  - use Ctrl+C to interrupt current work;
  - reset the device and try again;
  - check connection properties;
  - make sure the device has suitable MicroPython / CircuitPython /
firmware;
  - make sure the device is not in bootloader mode.

I did not seem to have this trouble before installing release 4.0.1, but if
I try the 3.3.14 version it connects and I can get the microPython prompt
after Stop/Restart backend, but it outputs: rst:0x1 (POWERON_RESET),boot:0x1
before I can do anything and I can't even open a file on the ESP32.

Kevin

bikingfreak, 2024-05-18:
Aivar,
I am trying to load & retrieve files from an ESP32 and both versions (4.1.4
and 3.3.14) will not let me stop the execution of the code on the ESP32 to
handle the files. This is on Ubuntu 22.04.4 on an hp Victus 15L Gaming
Desktop TG02-0xxx desktop.

Kevin