thonny/thonny #3184: Micropython debugging

thierryppp, 2024-05-02:
In micropython with Thonny IDE
Is it possible to import modules from a computer disk during test / debugging, but not from flash?

Example ==> in subdirectory /toto you have a file foo.py

Code 
 ```
import time
from machine import Pin, freq
from toto.foo import foo ==> error no subdirectory toto
```
It's normal on your card /flash you haven't sub toto

Thanks,
Thierry.

wohltat, 2024-05-10:
I don't get what's the problem with that.
Can't you just copy the file to the file system on the flash into the /toto/ folder?

As far as i know it is not possible to directly import files from disk. You have to either copy them or run a script (F5) and have them pasted.

wohltat, 2024-05-10:
Speaking of micropython debugging, it would be useful maybe to integrate debugging capabilities from the **mpdb** project into thonny.
https://github.com/bobveringa/mpdb

aivarannamaa, 2024-05-11:
@thierryppp, Thonny does not support loading modules from the development machine, but [mpremote](https://docs.micropython.org/en/latest/reference/mpremote.html) does, by using a clever way of mapping a special kind of file system in MicroPython.

@wohltat, it looks like mpdb does not offer any solutions to the main problem with MicroPython debugging -- the lack of access to local variables. You can read more at https://forum.micropython.org/viewtopic.php?t=7995