thonny/thonny #3185: Thonny Back End - Modules not found

KiwiWilkinson, 2024-05-03:
When I try to use Thonny features, I get an error.

When I try to debug I get:

PROBLEM IN THONNY'S BACK-END: Exception while handling 'Debug' (ModuleNotFoundError: No module named 'asttokens'). See Thonny's backend.log for more info.

If Syntax Highlighting is on, I get:

PROBLEM IN THONNY'S BACK-END: Exception while handling 'highlight_occurrences' (ModuleNotFoundError: No module named 'jedi'). See Thonny's backend.log for more info.

The backend.log is below.

This is on a Raspberry Pi Zero W, connected over WiFi using SSH.

12:34:31.412 [MainThread] INFO    thonny: Thonny version: 4.1.4
12:34:31.412 [MainThread] INFO    thonny: cwd: C:\Users\kiwi
12:34:31.412 [MainThread] INFO    thonny: original argv: ['C:\\Program Files (x86)\\Thonny\\python.exe', '-s', '-u', '-B', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\cpython_ssh\\cps_back.py', "{'host': 'raspberrypi', 'user': 'administrator', 'interpreter': 'python3', 'cwd': '/home/administrator/Bus-Departure-Board', 'main_backend_options': {'run.warn_module_shadowing': True}}"]
12:34:31.412 [MainThread] INFO    thonny: sys.executable: C:\Program Files (x86)\Thonny\python.exe
12:34:31.412 [MainThread] INFO    thonny: sys.argv: ['C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\cpython_ssh\\cps_back.py', "{'host': 'raspberrypi', 'user': 'administrator', 'interpreter': 'python3', 'cwd': '/home/administrator/Bus-Departure-Board', 'main_backend_options': {'run.warn_module_shadowing': True}}"]
12:34:31.412 [MainThread] INFO    thonny: sys.path: ['C:\\Program Files (x86)\\Thonny\\lib\\site-packages\\thonny\\plugins\\cpython_ssh', 'C:\\Program Files (x86)\\Thonny\\python310.zip', 'C:\\Program Files (x86)\\Thonny\\DLLs', 'C:\\Program Files (x86)\\Thonny\\lib', 'C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages']
12:34:31.412 [MainThread] INFO    thonny: sys.flags: sys.flags(debug=0, inspect=0, interactive=0, optimize=0, dont_write_bytecode=1, no_user_site=1, no_site=0, ignore_environment=0, verbose=0, bytes_warning=0, quiet=0, hash_randomization=1, isolated=0, dev_mode=False, utf8_mode=0, warn_default_encoding=0, int_max_str_digits=-1)
12:34:31.412 [MainThread] INFO    thonny.plugins.cpython_ssh.cps_back: Starting mediator for administrator @ raspberrypi
12:34:31.413 [MainThread] INFO    thonny.common: Using THONNY_FRONTEND_SYS_PATH ['C:\\Program Files (x86)\\Thonny\\python310.zip', 'C:\\Program Files (x86)\\Thonny\\DLLs', 'C:\\Program Files (x86)\\Thonny\\lib', 'C:\\Program Files (x86)\\Thonny', 'C:\\Program Files (x86)\\Thonny\\lib\\site-packages', 'C:\\Users\\kiwi\\AppData\\Roaming\\Thonny\\plugins\\Python310\\site-packages']
12:34:32.149 [MainThread] INFO    thonny.plugins.cpython_ssh.cps_back: Uploading main backend
12:34:32.550 [MainThread] INFO    thonny.plugins.cpython_ssh.cps_back: Starting remote process: ['python3', '/tmp/thonny-backend-4.1.4-administrator/thonny/plugins/cpython_backend/cp_launcher.py', '/home/administrator/Bus-Departure-Board', "{'run.warn_module_shadowing': True}"]
12:34:33.754 [MainThread] INFO    thonny.plugins.cpython_ssh.cps_back: Starting response forwarder
12:34:33.755 [MainThread] INFO    thonny: TIME/MODS 2.356 Beginning of mainloop (+212 modules)
12:34:33.755 [MainThread] INFO    thonny: NEW MODS ['__future__', '_bcrypt', '_bcrypt.lib', '_bisect', '_blake2', '_bz2', '_cffi_backend', '_compression', '_ctypes', '_datetime', '_hashlib', '_heapq', '_lzma', '_queue', '_random', '_sha512', '_socket', '_sodium', '_sodium.lib', '_struct', 'array', 'base64', 'bcrypt', 'bcrypt.__about__', 'bcrypt._bcrypt', 'binascii', 'bisect', 'bz2', 'calendar', 'cryptography', 'cryptography.__about__', 'cryptography.exceptions', 'cryptography.hazmat', 'cryptography.hazmat._oid', 'cryptography.hazmat.backends', 'cryptography.hazmat.backends.openssl', 'cryptography.hazmat.backends.openssl.aead', 'cryptography.hazmat.backends.openssl.backend', 'cryptography.hazmat.backends.openssl.ciphers', 'cryptography.hazmat.backends.openssl.cmac', 'cryptography.hazmat.backends.openssl.dh', 'cryptography.hazmat.backends.openssl.dsa', 'cryptography.hazmat.backends.openssl.ec', 'cryptography.hazmat.backends.openssl.ed25519', 'cryptography.hazmat.backends.openssl.ed448', 'cryptography.hazmat.backends.openssl.hashes', 'cryptography.hazmat.backends.openssl.hmac', 'cryptography.hazmat.backends.openssl.poly1305', 'cryptography.hazmat.backends.openssl.rsa', 'cryptography.hazmat.backends.openssl.utils', 'cryptography.hazmat.backends.openssl.x25519', 'cryptography.hazmat.backends.openssl.x448', 'cryptography.hazmat.bindings', 'cryptography.hazmat.bindings._openssl', 'cryptography.hazmat.bindings._openssl.lib', 'cryptography.hazmat.bindings._rust', 'cryptography.hazmat.bindings.openssl', 'cryptography.hazmat.bindings.openssl._conditional', 'cryptography.hazmat.bindings.openssl.binding', 'cryptography.hazmat.primitives', 'cryptography.hazmat.primitives._asymmetric', 'cryptography.hazmat.primitives._cipheralgorithm', 'cryptography.hazmat.primitives._serialization', 'cryptography.hazmat.primitives.asymmetric', 'cryptography.hazmat.primitives.asymmetric.dh', 'cryptography.hazmat.primitives.asymmetric.dsa', 'cryptography.hazmat.primitives.asymmetric.ec', 'cryptography.hazmat.primitives.asymmetric.ed25519', 'cryptography.hazmat.primitives.asymmetric.ed448', 'cryptography.hazmat.primitives.asymmetric.padding', 'cryptography.hazmat.primitives.asymmetric.rsa', 'cryptography.hazmat.primitives.asymmetric.types', 'cryptography.hazmat.primitives.asymmetric.utils', 'cryptography.hazmat.primitives.asymmetric.x25519', 'cryptography.hazmat.primitives.asymmetric.x448', 'cryptography.hazmat.primitives.ciphers', 'cryptography.hazmat.primitives.ciphers.algorithms', 'cryptography.hazmat.primitives.ciphers.base', 'cryptography.hazmat.primitives.ciphers.modes', 'cryptography.hazmat.primitives.constant_time', 'cryptography.hazmat.primitives.hashes', 'cryptography.hazmat.primitives.kdf', 'cryptography.hazmat.primitives.kdf.scrypt', 'cryptography.hazmat.primitives.serialization', 'cryptography.hazmat.primitives.serialization.base', 'cryptography.hazmat.primitives.serialization.pkcs12', 'cryptography.hazmat.primitives.serialization.pkcs7', 'cryptography.hazmat.primitives.serialization.ssh', 'cryptography.utils', 'cryptography.x509', 'cryptography.x509.base', 'cryptography.x509.certificate_transparency', 'cryptography.x509.extensions', 'cryptography.x509.general_name', 'cryptography.x509.name', 'cryptography.x509.oid', 'ctypes', 'ctypes._endian', 'ctypes.wintypes', 'datetime', 'email', 'email._parseaddr', 'email.base64mime', 'email.charset', 'email.encoders', 'email.errors', 'email.quoprimime', 'email.utils', 'encodings.ascii', 'encodings.idna', 'errno', 'faulthandler', 'fnmatch', 'getpass', 'hashlib', 'heapq', 'hmac', 'ipaddress', 'locale', 'lzma', 'math', 'msvcrt', 'nacl', 'nacl._sodium', 'nacl.bindings', 'nacl.bindings.crypto_aead', 'nacl.bindings.crypto_box', 'nacl.bindings.crypto_core', 'nacl.bindings.crypto_generichash', 'nacl.bindings.crypto_hash', 'nacl.bindings.crypto_kx', 'nacl.bindings.crypto_pwhash', 'nacl.bindings.crypto_scalarmult', 'nacl.bindings.crypto_secretbox', 'nacl.bindings.crypto_secretstream', 'nacl.bindings.crypto_shorthash', 'nacl.bindings.crypto_sign', 'nacl.bindings.randombytes', 'nacl.bindings.sodium_core', 'nacl.bindings.utils', 'nacl.encoding', 'nacl.exceptions', 'nacl.public', 'nacl.signing', 'nacl.utils', 'paramiko', 'paramiko._version', 'paramiko._winapi', 'paramiko.agent', 'paramiko.auth_handler', 'paramiko.auth_strategy', 'paramiko.ber', 'paramiko.buffered_pipe', 'paramiko.channel', 'paramiko.client', 'paramiko.common', 'paramiko.compress', 'paramiko.config', 'paramiko.dsskey', 'paramiko.ecdsakey', 'paramiko.ed25519key', 'paramiko.file', 'paramiko.hostkeys', 'paramiko.kex_curve25519', 'paramiko.kex_ecdh_nist', 'paramiko.kex_gex', 'paramiko.kex_group1', 'paramiko.kex_group14', 'paramiko.kex_group16', 'paramiko.kex_gss', 'paramiko.message', 'paramiko.packet', 'paramiko.pipe', 'paramiko.pkey', 'paramiko.primes', 'paramiko.proxy', 'paramiko.rsakey', 'paramiko.server', 'paramiko.sftp', 'paramiko.sftp_attr', 'paramiko.sftp_client', 'paramiko.sftp_file', 'paramiko.sftp_handle', 'paramiko.sftp_server', 'paramiko.sftp_si', 'paramiko.ssh_exception', 'paramiko.ssh_gss', 'paramiko.transport', 'paramiko.util', 'paramiko.win_openssh', 'paramiko.win_pageant', 'pathlib', 'platform', 'posixpath', 'queue', 'quopri', 'random', 'select', 'selectors', 'shlex', 'shutil', 'signal', 'socket', 'stringprep', 'struct', 'subprocess', 'tempfile', 'thonny.backend', 'unicodedata', 'urllib', 'urllib.parse', 'zlib']

aivarannamaa, 2024-05-11:
Please try installing `asttokens` and `jedi` also on your RPi. Thonny will send Python code to the remote device, which require those modules.

Damian-2001, 2024-05-14:
Hello! I have a similar error in thonny portable when I invoke calltips while debugging. The error description in Shell view is:
```
PROBLEM IN THONNY'S BACK-END: Exception while handling 'get_editor_calltip' (ModuleNotFoundError: No module named 'jedi').
See Thonny's backend.log for more info.
Traceback (most recent call last):
  File "D:\Archivos de programa\Thonny\lib\site-packages\thonny\backend.py", line 285, in _handle_normal_command
    response = handler(cmd)
  File "D:\Archivos de programa\Thonny\lib\site-packages\thonny\backend.py", line 400, in _cmd_get_editor_calltip
    from thonny import jedi_utils
  File "D:\Archivos de programa\Thonny\lib\site-packages\thonny\plugins\cpython_backend\cp_back.py", line 352, in _custom_import
    module = self._original_import(*args, **kw)
  File "D:\Archivos de programa\Thonny\lib\site-packages\thonny\jedi_utils.py", line 8, in <module>
    import jedi.api.classes
  File "D:\Archivos de programa\Thonny\lib\site-packages\thonny\plugins\cpython_backend\cp_back.py", line 352, in _custom_import
    module = self._original_import(*args, **kw)
ModuleNotFoundError: No module named 'jedi'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "D:\PythonSRC\youtube.py", line 209, in <module>
    url = input("Video url: ") or ""
  File "D:\Archivos de programa\Thonny\lib\site-packages\thonny\plugins\cpython_backend\cp_back.py", line 1207, in readline
    return self._generic_read("readline", limit)
  File "D:\Archivos de programa\Thonny\lib\site-packages\thonny\plugins\cpython_backend\cp_back.py", line 1192, in _generic_read
    self._backend._handle_normal_command(msg)
  File "D:\Archivos de programa\Thonny\lib\site-packages\thonny\plugins\cpython_backend\cp_back.py", line 272, in _handle_normal_command
    super()._handle_normal_command(cmd)
  File "D:\Archivos de programa\Thonny\lib\site-packages\thonny\backend.py", line 304, in _handle_normal_command
    sys.exit(1)
SystemExit: 1
```

Apparently, the error appears only when I change the interpreter to another python installation (without the Jedi module installed)

KiwiWilkinson, 2024-05-14:
> Please try installing `asttokens` and `jedi` also on your RPi. Thonny will send Python code to the remote device, which require those modules.

Thank you, that worked.

closed 2024-05-14 by KiwiWilkinson