thonny/thonny #3187: remote files are not opened when clicking on the stacktrace hyperlinks

wohltat, 2024-05-05:
When an error occurs in a python file for example boot.py, clicking on the stacktrace hyperlink for that file opens a local file instead of the remote one.
This is obviously know according to the TODO comments..
https://github.com/thonny/thonny/blob/78a6fa8b7accde063f753ae77d918ffe6dea9e25/thonny/shell.py#L1490-L1491

What about the following solution. It works on my machine. Tested it for micropython and local python.

``` python
from thonny.editors import make_remote_path

def _handle_hyperlink(self, event):
        import webbrowser

        try:
            line = self.get("insert linestart", "insert lineend")
            # Python stacktrace
            matches = list(re.finditer(r'File "(?P<file>[^"]+)", line (?P<line>\d+)', line))
            if not matches:
                # Friendly traceback
                matches = list(
                    re.finditer(
                        r"\b(?P<line>\d+)\b.+'(?P<file>[^\']+\.pyw?)'", line, flags=re.IGNORECASE
                    )
                )

            if len(matches) == 1:
                filename = os.path.expanduser(matches[0].group("file"))
                lineno = int(matches[0].group("line"))
                if (
                    filename in (STRING_PSEUDO_FILENAME, REPL_PSEUDO_FILENAME)
                    and self._last_main_file
                ):
                    filename = self._last_main_file

                # NB! Don't attempt to check the existence of the file as it may be remote file
                # or editor id (of untitled editor)
                # TODO: handle remote files
                # TODO: better use events instead direct referencing
                filename_prefix = ''
                if runner := get_runner():
                    if proxy := runner.get_backend_proxy():
                        from thonny.plugins.cpython_frontend import LocalCPythonProxy
                        if proxy.is_connected() and not isinstance(proxy, LocalCPythonProxy):
                            filename_prefix = make_remote_path('/')

                get_workbench().get_editor_notebook().show_file(filename_prefix + filename, lineno, set_focus=False)

            else:
                r = self.tag_prevrange("io_hyperlink", "@%d,%d" % (event.x, event.y))
                if r and len(r) == 2:
                    url = self.get(r[0], r[1])
                    if SIMPLE_URL_SPLIT_REGEX.match(url):
                        webbrowser.open(url)

        except Exception as e:
            logger.exception("Could not handle hyperlink click", exc_info=e)
```

Can you give an example of how this could be handled with events?

------------------

Thonny 5.0.0.b1-dev
Manjaro Linux
Python 3.11.8
Tk 8.6.14

wohltat, 2024-05-05:
The Code above still has problems with <stdin> and files in subfolders.
I'll try to fix it later.

aivarannamaa, 2024-05-11:
Thanks for the suggestion!

Actually it is not always clear, whether to open the file on the device or on the development machine. Some users keep and edit the master copies on the dev machine and upload when ready. Of course, Thonny could keep track of this relationship and open the master copy. I'll consider implementing this in the next feature release after 5.0.

wohltat, 2024-06-22:
> Thanks for the suggestion!
>
> Actually it is not always clear, whether to open the file on the device or on the development machine. Some users keep and edit the master copies on the dev machine and [...]

I think since there is no automated synchronization in place linking to the files on the device seems the best solution for now. Furthermore the links are very likely not referencing those master copies because the file paths are often in the root directory.