thonny/thonny #3189: Open system shell not working

poon64, 2024-05-08:
I get the error below when I try to open the system shell in Ubuntu 24.04 and Thonny 4.1.4. 
I'm using /usr/bin/gnome-terminal.wrapper as terminal

Traceback (most recent call last):
  File "/snap/thonny/217/lib/python3.10/tkinter/__init__.py", line 1921, in __call__
    return self.func(*args)
  File "/snap/thonny/217/lib/python3.10/site-packages/thonny/workbench.py", line 1109, in dispatch_from_menu
    dispatch(None)
  File "/snap/thonny/217/lib/python3.10/site-packages/thonny/workbench.py", line 1035, in dispatch
    handler()
  File "/snap/thonny/217/lib/python3.10/site-packages/thonny/plugins/system_shell/__init__.py", line 56, in _open_system_shell
    return terminal.run_in_terminal(cmd, cwd, env_overrides, True)
  File "/snap/thonny/217/lib/python3.10/site-packages/thonny/terminal.py", line 19, in run_in_terminal
    _run_in_terminal_in_linux(cmd, cwd, env, keep_open)
  File "/snap/thonny/217/lib/python3.10/site-packages/thonny/terminal.py", line 74, in _run_in_terminal_in_linux
    term_cmd = _get_linux_terminal_command()
  File "/snap/thonny/217/lib/python3.10/site-packages/thonny/terminal.py", line 208, in _get_linux_terminal_command
    raise RuntimeError("Don't know how to open terminal emulator")
RuntimeError: Don't know how to open terminal emulator

aivarannamaa, 2024-05-11:
Thonny has knowledge about most common terminal programs, so that it can open it with correct environment, working directory etc.

Can you tell me more about `/usr/bin/gnome-terminal.wrapper` -- is it similar to another termina in terms of its command line switches and capabilities? You can investigate https://github.com/thonny/thonny/blob/master/thonny/terminal.py to learn, which kind of details Thonny needs to take care of to support a terminal program on Linux.

sameersharma2006, 2024-05-24:
@aivarannamaa  @poon64  Snaps are sand-boxed & hence can not use system terminal, to get thonny to use system shell some terminator emulator will have to be shipped in the snap itself (Not ideal in any case).

The better option for thonny is to detect weather the host is snap by getting the base  OS as `Ubuntu core ...`  and adding some sort of `if` clause to produce better error message like for e.g. 'This option is not supported within snap sandbox'