thonny/thonny #3191: What is the meaning of this error message OSError: [Errno 1] EPERM in thonny?

ocmak2078, 2024-05-12:


coffandro, 2024-05-14:
According to some googling it seems to be a network error of some kind, could you provide further detail about what lead to this error and such?

ocmak2078, 2024-05-14:
I ran this program in a new raspberry pi picoW,  it stopped and display the
error message : OSError: {Errno 1] EPRRM.
However, the program runs OK with another raspberry pi picoW, may be the
new picoW which I bought recently, has wifi  connect problems?

wifi=network.WLAN(network.STA_IF)
time.sleep(.5)
wifi.active(True)
time.sleep(.5)
wifi.connect(secrets.ssid, secrets.password)

Cornelius Andreas Rosenaa ***@***.***> 於 2024年5月14日 週二
下午2:26寫道：

coffandro, 2024-05-14:
So the EXACT same app runs differently on different Picos? this sounds weird but no matter what dosen't quite seem like a Thonny problem imho

ocmak2078, 2024-05-14:
Agreed.  Thanks for your reply !

Cornelius Andreas Rosenaa ***@***.***> 於 2024年5月14日 週二
下午3:19寫道：

coffandro, 2024-05-31:
Since we both agree this is an issue with something other than thonny, how about you close the issue?

closed 2024-07-12 by aivarannamaa