thonny/thonny #3194: Save As doesn't work, no reaction

Libelldrian, 2024-05-14:
I installed Thonny on Ubuntu via $ sudo apt install thonny.

When I try to save a file for the first time using "save as", the program doesn't react - no window pops up.

aivarannamaa, 2024-05-14:
Please check, whether the file dialog has appeared behind the main window.

The work-around in this case is to check "Use Tk file dialogs instead of Zenity" in Tools => Options => General.

nbehrnd, 2024-05-16:
@aivarannamaa I came here for the same observation reported by @Libelldrian which is about Thonny (version 4.0.1 in Debian 13/trixie with Python 3.11.9 and Tk 8.6.14) in simultaneous presence of  zenity for gnome (version 4.0.1) -- both installed by the distribution's package manager `synaptic` and from Debian's repositories.  However, affirmative, the bypass to use Tk dialogs instead of zenity provides the wanted functionality.

JLuc, 2025-11-11:
Same here. It makes a bad first impression since creating a file is the first use step. 

Switching to Tk fixes the issue. Should be default ?

nbehrnd, 2025-11-11:
@JLuc May you provide some additional detail than "Same here"?  One reason is my comment in May 2024 was about Thonny version 4.0.1 in Linux Debian.  On the other hand, Thonny (version 4.1.7, i.e. the current release) in Windows 10 (I know ...) with Python 3.10.11 and Tk 8.6.13 provides a functional `Save as`.

nbehrnd, 2025-11-11:
@JLuc Affirmative regarding Ubuntu 24.04 LTS does not provide the current version.  Debian's package tracker, right hand column, features a section `other distros` to link to the (non-Debian) [record] on repology.org.

On the other hand, a virtual environment of Python would allow to run the installation via [PyPI].  Sometimes instead of an installation with `pip`, [pipx] is advantageous because it doesn't require a _manual_ reactivation of the virtual environment to access functionality of a Python module from the CLI.

[record]: https://repology.org/project/thonny/badges
[PyPI]: https://pypi.org/project/thonny/
[pipx]: https://pypi.org/project/pipx/