thonny/thonny #3197: Can't install local micropython type stubs

wohltat, 2024-05-17:
There are micropython stubs missing in Thonny like for example esp, esp32, ... . 
I tried to install them from local file but i get this error message:

```
/usr/bin/python -m pipkin --dir /home/boogieman/.config/Thonny/stubs/ESP32 install /home/username/build/micropython-stubs/stubs/micropython-v1_23_0_preview-esp32-ESP32_GENERIC-merged/esp32.pyi
Using PipkinProxy at http://127.0.0.1:36628
ERROR: Invalid requirement: '/home/username/build/micropython-stubs/stubs/micropython-v1_23_0_preview-esp32-ESP32_GENERIC-merged/esp32.pyi'
Hint: It looks like a path. The path does exist. 
Error: process returned with code 1
```

wohltat, 2024-05-17:
I just copied them manually into `thonny/thonny/plugins/micropython/base_api_stubs` and it seems to work right away without restarting.

Took the stubs from https://github.com/Josverl/micropython-stubs.

aivarannamaa, 2024-05-17:
Thonny 5.0 will allow installing them in the configuration dialog (#2186)

closed 2024-05-17 by aivarannamaa

Witawat, 2024-12-20:
> I just copied them manually into `thonny/thonny/plugins/micropython/base_api_stubs` and it seems to work right away without restarting.
> 
> Took the stubs from https://github.com/Josverl/micropython-stubs.

how to ?

i try copy

[image]

to  AppData\Local\Programs\Thonny\Lib\site-packages\thonny\plugins\esp\esp32_api_stubs

not work

[image]