thonny/thonny #3199: Cannot connect in SSH

Sill1000, 2024-05-18:
Hi,

Thanks for this nice Software

Unfortunately I canot connect my raspberry pi in SSH.

`Can't connect to 'xxx.xxx.x.xxx' with user 'test': Authentication failed.
Re-check your host, authentication method, password or keys.

Process ended with exit code 1.`

But from Terminal (MAC) 
after
ssh test
it asks the pw. And it works.

Of course, I did a copy paste form the pw to be sure !

On the Pi it's Python 3.11.2

Why do I wrong ?

Thanks

aivarannamaa, 2024-05-22:
You said `ssh test` works for you on Mac. This gives me the impression that `test` is the remote host name, but `Can't connect to 'xxx.xxx.x.xxx' with user 'test'` implies `test` is the user. Which is it?

Sill1000, 2024-05-23:
Thanks for your answers ! Probably my fault but it’s not clear for me:

With Terminal I do SSH test and after that I enter my password and it works.

With Thonny I put my Raspberry IP address into Host
Username:
 test
Authentification password.

What is wrong ?

Thanks

Sill1000, 2024-05-23:
Problem founded ! Sorry I was mistaken on the username ! Sorry ! And thanks agains

aivarannamaa, 2024-07-12:
The backend.log can be found by selecting "Tools => Open Thonny data folder".

The error message means you should install Python package named `jedi` for your target Python.

salbertini, 2025-02-02:
Hello aivarannamaa,
thanks for all the suppurt!!

I've more or less the same problem, but I'm not at all an expert of Raspberry PI.
I follow your suggest (in another post) to connect via RPI with a remote python 3 SSH.
I insert the IP of the host, the username and authentication wit password.
I insert the password but this error message appear "Can't connect to 'xxx.xxx.x.xxx ' with user 'albe': [Errno 11001] getaddrinfo failed
Re-check your host, authentication method, password or keys.

Process ended with exit code 1."

Usually I use Putty to connect to my RPI in headless mode, and I use the same user and password as tried for Thonny.

Probably my RPI is too hold? I'm using Rasperry PI 1B with Rasperry PI OS.
thanks
Stefano

aivarannamaa, 2025-02-04:
@salbertini, in the error message you have extra space after the x-s. Perhaps this is the cause?

salbertini, 2025-02-04:
@aivarannamaa no it's not the cause I've tryed to retype and same error.
Anyway, I've ordered a new PI 5, it should arrive in a couple of week. Let me try with the new Hardware, I'll come back to you.
Thanks

salbertini, 2025-02-09:
I've received PI 5 and Thonny works perfectly via SSH. So probably the issue with the old RP is due to alder HW.
Thanks

aivarannamaa, 2025-02-10:
Great!

closed 2025-02-10 by aivarannamaa